ethz-asl/ethz_piksi_ros #5: Upgrade Piksi Multi Firmware from 1.2 to 1.3

marco-tranzatto, 2018-01-08:
> Firmware Version 1.3 now provides:
> - Improved RTK Fixed Mode Reliability
> - Improved L2C Tracking Capability
> - New UDP Client and Servers
> - Raw Magnetometer Output
>
> For more detailed information, refer to the [Piksi Multi Firmware 1.3 Release](https://swiftnav.us8.list-manage.com/track/click?u=0085d053dc5c79d1e62f397e4&id=cbe9492105&e=c5b9c0e86d)

matteopantano, 2018-03-20:
Hi Marco,

Can I use  the piksi firmware 1.4.10 with LIB SBP 2.3.10? Or should I downgrade? Seems that Swiftnav released 1.4.10 to fix a major problem with satellites

marco-tranzatto, 2018-03-20:
Hi Matteo,

an update of libsbp is under development at the moment in another branch. We haven't finished it yet and would need to test it. You can try to revert to what master has been tested with (Firmware 1.2.14, rembmer to run the install script to install the correct version of libsbp) but I guess you will get issue with GLONASS.

Next time please open another issue instead of using an existing one not directly related to your topic.

Regards,
Marco.

marco-tranzatto, 2018-03-28:
Addressed in PR #16

closed 2018-03-28 by marco-tranzatto