ethz-asl/ethz_piksi_ros #29: Cannot install libsbp

didriksg, 2018-05-01:
When trying to install the libsbp, using the `install_piksi_multi.sh`, everything seems to be working fine. I have added the output from the 

` 
Installing SBP library for Piksi Multi.
Cloning into 'libsbp'...
remote: Counting objects: 15786, done.
remote: Compressing objects: 100% (53/53), done.
remote: Total 15786 (delta 14), reused 23 (delta 6), pack-reused 15719
Receiving objects: 100% (15786/15786), 20.75 MiB | 240.00 KiB/s, done.
Resolving deltas: 100% (10758/10758), done.
Checking connectivity... done.
Note: checking out 'v2.3.10'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at fb67f2d... Merge pull request #549 from denniszollo/v2.3.10_branch
 
Another version of libsbp is already installed in the system.
Do you wish to remove it and install version v2.3.10 ? [y or Y to accept]
y
Removing previous libsbp.
Installing SBP dependencies.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
pandoc is already the newest version (1.16.0.2~dfsg-1).
The following packages were automatically installed and are no longer required:
  linux-headers-4.13.0-36 linux-headers-4.13.0-36-generic linux-headers-4.13.0-37 linux-headers-4.13.0-37-generic
  linux-image-4.13.0-36-generic linux-image-4.13.0-37-generic linux-image-extra-4.13.0-36-generic linux-image-extra-4.13.0-37-generic
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 79 not upgraded.
Requirement already satisfied: tox in /usr/local/lib/python3.5/dist-packages (3.0.0)
Requirement already satisfied: pluggy<1.0,>=0.3.0 in /usr/local/lib/python3.5/dist-packages (from tox) (0.6.0)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from tox) (1.10.0)
Requirement already satisfied: py>=1.4.17 in /usr/local/lib/python3.5/dist-packages (from tox) (1.5.3)
Requirement already satisfied: virtualenv>=1.11.2 in /usr/local/lib/python3.5/dist-packages (from tox) (15.2.0)
Requirement already satisfied: construct==2.9.33 in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 1)) (2.9.33)
Requirement already satisfied: futures>=2.2.0 in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 2)) (3.1.1)
Requirement already satisfied: httpretty in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 3)) (0.9.4)
Requirement already satisfied: pyftdi==0.13.4 in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 4)) (0.13.4)
Requirement already satisfied: pylibftdi in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 5)) (0.17.0)
Requirement already satisfied: pyserial in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 6)) (3.4)
Requirement already satisfied: requests>=2.8.1 in /usr/lib/python3/dist-packages (from -r requirements.txt (line 7)) (2.9.1)
Requirement already satisfied: requests-futures>=0.9.5 in /usr/local/lib/python3.5/dist-packages (from -r requirements.txt (line 8)) (0.9.7)
Requirement already satisfied: six in /usr/lib/python3/dist-packages (from httpretty->-r requirements.txt (line 3)) (1.10.0)
Requirement already satisfied: pyusb>=1.0.0b1 in /usr/local/lib/python3.5/dist-packages (from pyftdi==0.13.4->-r requirements.txt (line 4)) (1.0.2)
Requirement already satisfied: markupsafe in /usr/lib/python3/dist-packages (0.23)
running install
running bdist_egg
running egg_info
creating sbp.egg-info
writing requirements to sbp.egg-info/requires.txt
writing sbp.egg-info/PKG-INFO
writing top-level names to sbp.egg-info/top_level.txt
writing dependency_links to sbp.egg-info/dependency_links.txt
writing manifest file 'sbp.egg-info/SOURCES.txt'
reading manifest file 'sbp.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
warning: no files found matching '*.md'
warning: no files found matching '*.sh'
warning: no files found matching 'LICENSE'
warning: no files found matching '*.py' under directory 'sbp/'
no previously-included directories found matching 'docs/_build'
[... 353 lines cut ...]
 * /piksi/var_spp: [25.0, 25.0, 64.0]
 * /rosdistro: kinetic
 * /rosversion: 1.12.13

NODES
  /
    piksi (piksi_multi_rtk_ros/piksi_multi.py)

ROS_MASTER_URI=http://localhost:11311

process[piksi-1]: started with pid [6031]
[INFO] [1525183047.623463]: /piksi start
[FATAL] [1525183048.069252]: No SBP library found. Please install it by using script in 'install' folder.
[INFO] [1525183048.179455]: libsbp version currently used: -1
[WARN] [1525183048.179701]: Lib SBP version in usage (-1) is different than the one used to test this driver (2.3.10)!
Please run the install script: 'install/install_piksi_multi.sh'
[INFO] [1525183048.185284]: Swift driver started in normal mode.
[INFO] [1525183048.224338]: Starting device in Base Station Mode
Host Side Unhandled byte: 0xde
[piksi-1] process has finished cleanly
`

I have in the past used the `swiftnav_ros` package and installed the `libsbp` that that package required. Can that maybe be the case? How do I move forward for eventually removing that?

I have already configed my RTK card to send at the correct baudrate, and set it up as a base station.

didriksg, 2018-05-01:
I don't know what I did different, but it somehow works now... So it is not a big problem for me anymore, but you should definitely look into it

marco-tranzatto, 2018-05-02:
Hi,

if the error is not reproducible (I haven't seen this issue before) and you solved it by giving the install script another try then it's quite difficult for us to have a look at it.

Anyway, I'm glad that now works in your setup ;)

Regards,
Marco.

closed 2018-05-02 by marco-tranzatto