ethz-asl/ethz_piksi_ros #31: IMU data

matteopantano, 2018-05-08:
Dear Marco,

I am trying to use data coming from piksi in the topic `/piksi/imu_raw`. Swiftnav support says that the unit of measurements coming in this topic are related to what specified in the `imu_aux` field. Is there any way of reading this?

Thanks,

Matteo

marco-tranzatto, 2018-05-09:
Hi Matteo,

I think they are referring to these settings:

[image]

I think I did not understand your question about "imu_aux field". Where do you want to read it? You can either set the max range in the piksi console, or you can use the already existing functions to read a parameter using our ROS driver. Have a look [here](https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_rtk_ros/src/geodetic_survey.py#L154).

At the moment we do not read any magnetometer data. You can implement your own callback function, in a similar way as it's done for the IMU. I'm happy to review your pull request once you have a working solution.

Regards,
Marco.

matteopantano, 2018-05-13:
Thanks Marco!

Furthermore I did the integration with the magnetometer. Can I have the permission to push a branch with my suggested changes?

Regards,

Matteo

marco-tranzatto, 2018-05-13:
Hi Matteo,

you can fork this repo, do your changes and then open a Pull request directly from your fork. In this way you don't need write access to this repo. I will be happy to review the request and, if no changes are needed, we can merge it into master.

Cheers,
Marco.

matteopantano, 2018-05-15:
Ok I did not saw that, I already did the implementation on my own. Choose the best mag implementation ;)

marco-tranzatto, 2018-05-17:
Mag data added with PR #35

closed 2018-05-17 by marco-tranzatto