ethz-asl/ethz_piksi_ros #33: Nothing published in rtk float

Wagdi-Benyaala, 2018-05-14:
Hello,

First of all, I want to thank you for this elegant package ! I have actually written my own package to use TCP communication instead of serial. But well I got to admit that my package was not as elegant :).

I have been investigating the reason why i do not get anything published on the topic _'/piksi/navsatfix_best_fix'_ when I was in RTK Float mode. 

It actually seems that it is a wanted behavior and that RTK Float is only available in debug mode. However, is it also wanted that nothing is published on the _'/piksi/navsatfix_best_fix'_ topic ? I am actually using this topic as the main topic to get first SPP then RTK Float and finally RTK Fix. Should I use another topic ?

Also, I might need to post this part in another issue tell me if I should, is there a reason to use fix variances for spp, rtk float and rtk fix ? The message _MsgPosLLH_ provides a horizontal and vertical accuracy (in millimeters). In previous packages the square of those values where used to compute the three variances. 

Best,

marco-tranzatto, 2018-05-14:
Hi,

I'm happy you find our driver useful. After a recent pull request from SwiftNav a TCP interface is available as well. Check the parameter [interface](https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_rtk_ros/cfg/piksi_multi_driver_settings.yaml#L3) in the confg file.

About your question: yes, RTK float measurements are not published if debug mode is not especially enabled. This choice was made to limit the amount of advertised topics by the driver. You can change the debug mode [here](https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_rtk_ros/cfg/piksi_multi_driver_settings.yaml#L12). At the moment if debug mode is not enabled, then RTK float is NOT published at all in the `/piksi/navsatfix_best_fix` topic. This other decision was made when RTK float was not giving a good enough position information, therefore we decided not to publish this info. I haven't had the chance to compare RTK float and SPP, so I cannot really tell which one should be published as best available position measurement in case no RTK FIX is present.
So to answer directly to your question: you can access RTK float information only if debug_mode is enabled.

About your second question: probably you should open a new issue, however covariances for GPS measurements are quite trick to compute from measurements. We believe it is better to use fixed covariances for the three different modes. However you can use the DOP and HOP out from the driver to get an idea about the "covariances" of the measurements. Does that make sense for you?

Regards,
Marco.

marco-tranzatto, 2018-05-15:
Great! I'm glad I could help ;)

Marco.

closed 2018-05-15 by marco-tranzatto