ethz-asl/ethz_piksi_ros #56: Update for firmware v2.0

matteopantano, 2018-09-08:
Will be there an update to the repository for firmware v2.0 of piksi multi?

kholtmann, 2018-09-08:
Eventually there might be. We did not have time to work on it yet.

matteopantano, 2018-09-12:
Ok, thanks

matteopantano, 2019-01-04:
Hey @rikba I am using the Feature/v2.1.14. I had forked the repo and I have had several issues with the installation, however the major one was that the install script does not remove the old libsbp located in `/usr/local/lib/python2.7/dist-packages/sbp-*`. Did you encounter the same problem?

rikba, 2019-01-07:
Hi @matteopantano, 

I don't remember having issues running the install script. But I think I installed on a clean machine. 

Were you finally able to install and test the new version?

matteopantano, 2019-01-07:
Yes I have manually deleted old libsbp and now I am using the driver. If you want I can open a fork so you can evaluate the modifications that I did

kinergy, 2019-01-26:
don't think we had an issue with this - we switched to branch Feature/v2.1.14 and used the install script as is. I see that `sbp-2.4.1-py2.7.egg` is the only sbp- present.

marco-tranzatto, 2019-02-25:
Firmware updated to 2.1.14 after PR #63

closed 2019-02-25 by marco-tranzatto