ethz-asl/ethz_piksi_ros #65: Using swiftnav Duro without a base

GrantDare, 2019-01-17:
It is unclear if this library can support the [swiftnav Duro](https://www.swiftnav.com/duro) singularly without a base. For the application I require DD position data globally referenced. The[ old library](http://wiki.ros.org/swiftnav_ros) no longer compiles and directs to this repository. Could you please direct me, if it is possible using this library, to how to use such a module without using a base?

marco-tranzatto, 2019-01-21:
Hi,

we have never tried this ROS driver with DURO because we use only the Multi version in our lab. Have you tried to use it in any case? The rover launch file should work even without a base station.

Regards,
Marco.

marco-tranzatto, 2019-05-31:
closing due to inactivity

closed 2019-05-31 by marco-tranzatto