ethz-asl/ethz_piksi_ros #67: unused dependency on zope

paulharter, 2019-01-25:
Hi

piksi_multi_rtk_ros/src/piksi_multi_rtk_ros/piksi_multi.py

Has in unused import of a zope exception

`from zope.interface.exceptions import Invalid`

I cant see anywhere that zope is being used

marco-tranzatto, 2019-01-25:
Hi, thanks for reporting the issue.
I guess this is some legacy code left there by mistake. Feel free to open a pull request if you feel like fixing it :)

Thanks,
Marco.

mcburce, 2019-03-25:
Hi,
Thank you very much for developing this package.
We tried your package in our Husky UGV robot running in Ubuntu 16.04 LTS and ROS Kinetic. Everything seems okay at first, however, when we conducted tests in a large open field, we see some dead spots in the rtk fix topics of our rover gps in some locations. Kindly see attached image. The image on the left are the tracks/segments our robot followed while the image on the right shows the status of the piksi topics were were observing using rqt_bag. 

We also investigated using SwiftConsole instead of running piksi_multi_rover.launch. We moved our rover gps to those dead spot locations and traced the same tracks/segments our robot did, but we found no errors in the logged readings, we always get a solid rtk fix solution using SwiftConsole in all the tracks/segments. 

Can you please explain why these dead spots happen or occur if we run piksi_mulit_rover.launch?

[image]

Thanks a lot.

marco-tranzatto, 2019-03-26:
Hi @mcburce , please open another issue with your problem, as this is not really related to the topic of this issues.

Regards,
Marco.

mcburce, 2019-03-26:
Hi Marco,
Sorry about that, it's my first time posting an issue.Thanks for the advice.
Regards,@mcburce

marco-tranzatto, 2019-05-31:
closing due to inactivity

closed 2019-05-31 by marco-tranzatto