ethz-asl/ethz_piksi_ros #77: frequent drops in rtk fix solutions in field trials

mcburce, 2019-03-26:
Hi,
Thank you very much for developing this package.

We tried your package in our Husky UGV robot running in Ubuntu 16.04 LTS and ROS Kinetic. Everything seems okay at first, however, when we conducted tests in an open field (~250x150 m), we see some dead spots in the rtk fix topics of our rover gps in some locations. Kindly see attached image. The image on the left are the tracks/segments our robot followed in the field, while the image on the right shows the status of the piksi topics that we were observing using rqt_bag.

[image]

We also investigated using SwiftConsole instead of running piksi_multi_rover.launch. We moved our rover gps to those dead spot locations and traced the same tracks/segments our robot did, but we found no errors in the logged readings, we always get a solid rtk fix solution using SwiftConsole in all the tracks/segments.

Can you please explain why these dead spots happen or occur if we run piksi_mulit_rover.launch? How can we resolve it? Our robot's navigation is affected if for example one or more of its waypoint has no rtk fix signal.

Thanks a lot.

marco-tranzatto, 2019-04-01:
Hi @mcburce 

I haven't used Piksi multi lately, so I can't really say if this is a new issue.
@rikba have you seen something similar during your last tests?

mcburce, 2019-04-08:
Hi @marco-tranzatto 

Thank you very much for the reply. I hope @rikba will reply soon too.

rikba, 2019-04-15:
Hi @mcburce 

We couldnt observe RTK drops recently on our system but it can have several reasons why the RTK fix drops.

1. bad satellite signals
2. bad correction link
3. noise from other sensors that interferes with the GPS signal

> We moved our rover gps to those dead spot locations and traced the same tracks/segments our robot did
Did you do this on the same device at the same time or sequentially? Because if you did this sequentially it's hard to tell what the error source is.

What makes me a bit curious though is the complete drop of debug messages from 01:30:00 to 02:30:00 and 20:00:00 to 30:00:00. Did you turn off the driver during this time or did it not output messages anymore?

Have a look at our [new debug tools](https://github.com/ethz-asl/ethz_piksi_ros/pull/75) to debug corrections.
```
roslaunch roslaunch rqt_gps_rtk_plugin gui.launch
```
Also with the newest version and an L1/L2 receiver you should have about 30 or more satellites for the correction at any time.

Also have a look at the [postprocessing possibilities](https://github.com/ethz-asl/ethz_piksi_ros/wiki/Data-collection-and-post-processing) in case you do not need the RTK position online.

rikba, 2019-06-14:
Closing due to inactivity.

closed 2019-06-14 by rikba