ethz-asl/ethz_piksi_ros #79: No SBP library found. Please install it by using script in 'install' folder.

onatkeser, 2019-05-07:
Hi, as "didriksg" mentioned before, I have also installed the libsdbp using the install_piksi_multi.sh and everything was successful. Then, while starting a launch file I get this error:

```
/usr/local/lib/python2.7/dist-packages/requests/__init__.py:91: RequestsDependencyWarning: urllib3 (1.24.3) or chardet (2.3.0) doesn't match a supported version!
  RequestsDependencyWarning)
/usr/local/lib/python2.7/dist-packages/requests/__init__.py:83: RequestsDependencyWarning: Old version of cryptography ([1, 2, 3]) may cause slowdown.
  warnings.warn(warning, RequestsDependencyWarning)
[INFO] [1557217710.559212]: /piksi start
[FATAL] [1557217710.855365]: No SBP library found. Please install it by using script in 'install' folder.
Traceback (most recent call last):
  File "/home/can/swift_ros_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/bin/piksi_multi", line 6, in <module>
    piksi_multi = PiksiMulti()
  File "/home/can/swift_ros_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/src/piksi_multi_rtk_ros/piksi_multi.py", line 76, in __init__
    rospy.loginfo("libsbp version currently used: " + installed_sbp_version)
TypeError: cannot concatenate 'str' and 'int' objects
[piksi-2] process has died [pid 7200, exit code 1, cmd /home/can/swift_ros_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/bin/piksi_multi __name:=piksi __log:=/home/can/.ros/log/17512c86-70a2-11e9-9be0-e82aea8d73a9/piksi-2.log].
log file: /home/can/.ros/log/17512c86-70a2-11e9-9be0-e82aea8d73a9/piksi-2*.log
[piksi-2] restarting process
```
I retried the installation several times but nothing changed even on another computer. Could you please give suggestion? I guess many people had this issue and solved but no one knows how.. Thanks in advance

marco-tranzatto, 2019-05-07:
Hi, can you copy and paste the output of  the command `source install/install_piksi_multi.sh` ?

onatkeser, 2019-05-07:
Here it is,

```
~/swift_ros_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/install   master  ./install_piksi_multi.sh
 
Installing SBP library for Piksi Multi.
Cloning into 'libsbp'...
remote: Enumerating objects: 127, done.
remote: Counting objects: 100% (127/127), done.
remote: Compressing objects: 100% (77/77), done.
remote: Total 21456 (delta 59), reused 62 (delta 34), pack-reused 21329
Receiving objects: 100% (21456/21456), 88.88 MiB | 479.00 KiB/s, done.
Resolving deltas: 100% (14713/14713), done.
Checking connectivity... done.
Note: checking out 'v2.4.1'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at 58442f9... v2.4.1 files
 
Another version of libsbp is already installed in the system.
Do you wish to remove it and install version v2.4.1 ? [y or Y to accept]
Removing previous libsbp.
Installing SBP dependencies.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
pandoc is already the newest version (1.16.0.2~dfsg-1).
The following packages were automatically installed and are no longer required:
  fltk1.3-doc fluid fonts-lato gir1.2-gtk-2.0 hddtemp libassimp-dev libassimp3v5 libatk1.0-dev libcollada-dom2.4-dp-dev libcollada-dom2.4-dp0 libfltk-cairo1.3 libfltk-forms1.3 libfltk-gl1.3
  libfltk-images1.3 libfltk1.3 libfltk1.3-dev libidl-2-0 libinput-dev libjs-jquery-ui libkeybinder0 libllvm5.0 liborbit2 libpng16-16 libpyside-py3-2.0 libpyside2-dev libpyside2.0
  libqt5multimediaquick-p5 libqt5script5 libqt5scripttools5 libqt5svg5-dev libqt5x11extras5 libqt5x11extras5-dev libqt5xmlpatterns5 libqt5xmlpatterns5-dev libqt5xmlpatterns5-private-dev libruby2.3
  libshiboken-py3-2.0 libshiboken2-dev libshiboken2.0 libudev-dev liburdfdom-dev liburdfdom-headers-dev liburdfdom-model-state0.4 liburdfdom-model0.4 liburdfdom-sensor0.4 liburdfdom-tools
  liburdfdom-world0.4 libvte-common libvte9 libxcomposite-dev libxml2-utils libyaml-cpp-dev libyaml-cpp0.5v5 pyqt5-dev python-cycler python-gconf python-gnome2 python-imaging python-keybinder
  python-matplotlib python-matplotlib-data python-notify python-opengl python-pydot python-pyorbit python-pyqt5 python-pyqt5.qtopengl python-pyqt5.qtsvg python-pyqt5.qtwebkit python-pyside2
  python-pyside2.qtconcurrent python-pyside2.qtcore python-pyside2.qtgui python-pyside2.qthelp python-pyside2.qtnetwork python-pyside2.qtprintsupport python-pyside2.qtqml python-pyside2.qtquick
  python-pyside2.qtquickwidgets python-pyside2.qtscript python-pyside2.qtsql python-pyside2.qtsvg python-pyside2.qttest python-pyside2.qtuitools python-pyside2.qtwebkit python-pyside2.qtwebkitwidgets
  python-pyside2.qtwidgets python-pyside2.qtx11extras python-pyside2.qtxml python-sip python-sip-dev python-tz python-vte qtbase5-private-dev qtdeclarative5-private-dev qtmultimedia5-dev qtscript5-dev
  qtscript5-private-dev qttools5-dev-tools rake ros-kinetic-actionlib-tutorials ros-kinetic-angles ros-kinetic-bond-core ros-kinetic-bondpy ros-kinetic-camera-calibration-parsers
  ros-kinetic-camera-info-manager ros-kinetic-collada-parser ros-kinetic-collada-urdf ros-kinetic-common-msgs ros-kinetic-common-tutorials ros-kinetic-control-msgs ros-kinetic-diagnostic-aggregator
  ros-kinetic-diagnostic-analysis ros-kinetic-diagnostic-common-diagnostics ros-kinetic-diagnostics ros-kinetic-eigen-conversions ros-kinetic-eigen-stl-containers ros-kinetic-executive-smach
  ros-kinetic-filters ros-kinetic-geometric-shapes ros-kinetic-geometry ros-kinetic-geometry-tutorials ros-kinetic-gl-dependency ros-kinetic-image-common ros-kinetic-image-transport
  ros-kinetic-interactive-marker-tutorials ros-kinetic-interactive-markers ros-kinetic-joint-state-publisher ros-kinetic-kdl-conversions ros-kinetic-kdl-parser ros-kinetic-laser-assembler
  ros-kinetic-laser-filters ros-kinetic-laser-geometry ros-kinetic-laser-pipeline ros-kinetic-librviz-tutorial ros-kinetic-map-msgs ros-kinetic-media-export ros-kinetic-mk ros-kinetic-nav-msgs
  ros-kinetic-nodelet-core ros-kinetic-nodelet-tutorial-math ros-kinetic-octomap ros-kinetic-orocos-kdl ros-kinetic-pluginlib-tutorials ros-kinetic-polled-camera ros-kinetic-python-orocos-kdl
  ros-kinetic-python-qt-binding ros-kinetic-qt-dotgraph ros-kinetic-qt-gui ros-kinetic-qt-gui-cpp ros-kinetic-qt-gui-py-common ros-kinetic-qwt-dependency ros-kinetic-random-numbers
  ros-kinetic-resource-retriever ros-kinetic-robot-model ros-kinetic-robot-state-publisher ros-kinetic-ros ros-kinetic-ros-comm ros-kinetic-ros-tutorials ros-kinetic-rosbash ros-kinetic-rosboost-cfg
  ros-kinetic-rosconsole-bridge ros-kinetic-roscpp-core ros-kinetic-roscpp-tutorials ros-kinetic-roscreate ros-kinetic-roslang ros-kinetic-roslint ros-kinetic-roslisp ros-kinetic-rosmake
  ros-kinetic-rospy-tutorials ros-kinetic-rqt-action ros-kinetic-rqt-bag ros-kinetic-rqt-bag-plugins ros-kinetic-rqt-console ros-kinetic-rqt-dep ros-kinetic-rqt-graph ros-kinetic-rqt-gui
  ros-kinetic-rqt-gui-cpp ros-kinetic-rqt-gui-py ros-kinetic-rqt-launch ros-kinetic-rqt-logger-level ros-kinetic-rqt-moveit ros-kinetic-rqt-msg ros-kinetic-rqt-nav-view ros-kinetic-rqt-plot
[... cut ...]
_________________________________________________________________________________________________ summary __________________________________________________________________________________________________
  py27: commands succeeded
  congratulations :)

Finished running Python tests

Removing temporary folder.
SBP Library Installed
User can is already included in 'dialout' group.
 
Installation completed.

```

I had followed the instructions as in here https://support.swiftnav.com/customer/en/portal/articles/2924342-using-ros-with-swift-navigation-gnss-devices

marco-tranzatto, 2019-05-07:
Can you try to[ comment this line](https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_rtk_ros/src/piksi_multi_rtk_ros/piksi_multi.py#L74) and try again to launch it?

abhijeetknayak, 2019-09-14:
I face the same issue even though I have installed sbp and also find it in the Python2.7 folder. Any solution?

rikba, 2019-10-03:
We have simplified the [libsbp installation](https://github.com/ethz-asl/ethz_piksi_ros/blob/v1.11.0/piksi_multi_rtk_ros/install/install_piksi_multi.sh). Please try the new version.

closed 2019-10-03 by rikba

abhijeetknayak, 2019-10-04:
> We have simplified the [libsbp installation](https://github.com/ethz-asl/ethz_piksi_ros/blob/v1.11.0/piksi_multi_rtk_ros/install/install_piksi_multi.sh). Please try the new version.
Thank you for your reply. I dont know what the issue was, but SBP was installed in the system and the python script just wasn't able to find it. So, I just disabled lines of code in the script that search for the SBP library and it works fine.