ethz-asl/ethz_piksi_ros #83: No SBP library found, Although install script was excuted

MomoTavas, 2019-06-06:
Hello,

Although I ran the installation script located at ethz_piksi_ros/piksi_multi_rtk_ros/install/, roslaunch I get the following error. I also attached a .zip file of useful logs.

Thanks!

---------------------------------------------------------------------------------------------------------------
roslaunch piksi_multi_rtk_ros piksi_multi_base_station.launch 
... logging to /home/ocp-ais/.ros/log/33abc9f6-8685-11e9-927d-00bb609c48fc/roslaunch-OPC-AIS-15426.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://OPC-AIS:43529/

SUMMARY
========

PARAMETERS
 * /piksi_multi_base_station/base_station_ip_for_latency_estimation: 192.168.0.1
 * /piksi_multi_base_station/base_station_mode: True
 * /piksi_multi_base_station/baud_rate: 230400
 * /piksi_multi_base_station/broadcast_addr: 192.168.0.255
 * /piksi_multi_base_station/broadcast_port: 26078
 * /piksi_multi_base_station/debug_mode: False
 * /piksi_multi_base_station/driver_verbose: True
 * /piksi_multi_base_station/interface: serial
 * /piksi_multi_base_station/navsatfix_frame_id: base_link
 * /piksi_multi_base_station/publish_raw_imu_and_mag: False
 * /piksi_multi_base_station/serial_port: /dev/ttyUSB0
 * /piksi_multi_base_station/tcp_addr: 192.168.0.222
 * /piksi_multi_base_station/tcp_port: 55555
 * /piksi_multi_base_station/use_gps_time: False
 * /piksi_multi_base_station/var_rtk_fix: [0.0049, 0.0049, ...
 * /piksi_multi_base_station/var_rtk_float: [25.0, 25.0, 64.0]
 * /piksi_multi_base_station/var_spp: [25.0, 25.0, 64.0]
 * /piksi_multi_base_station/var_spp_sbas: [1.0, 1.0, 1.0]
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    piksi_multi_base_station (piksi_multi_rtk_ros/piksi_multi)

auto-starting new master
process[master]: started with pid [15436]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 33abc9f6-8685-11e9-927d-00bb609c48fc
process[rosout-1]: started with pid [15449]
started core service [/rosout]
process[piksi_multi_base_station-2]: started with pid [15466]
[INFO] [1559624228.488320]: /piksi_multi_base_station start
Traceback (most recent call last):
  File "/home/ocp-ais/.local/bin/pip", line 7, in <module>
    from pip._internal import main
ImportError: No module named _internal
[FATAL] [1559624228.920052]: No SBP library found. Please install it by using script in 'install' folder.
Traceback (most recent call last):
  File "/home/ocp-ais/swift_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/bin/piksi_multi", line 6, in <module>
    piksi_multi = PiksiMulti()
  File "/home/ocp-ais/swift_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/src/piksi_multi_rtk_ros/piksi_multi.py", line 89, in __init__
    rospy.loginfo("libsbp version currently used: " + installed_sbp_version)
TypeError: cannot concatenate 'str' and 'int' objects
[piksi_multi_base_station-2] process has died [pid 15466, exit code 1, cmd /home/ocp-ais/swift_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/bin/piksi_multi __name:=piksi_multi_base_station __log:=/home/ocp-ais/.ros/log/33abc9f6-8685-11e9-927d-00bb609c48fc/piksi_multi_base_station-2.log].
log file: /home/ocp-ais/.ros/log/33abc9f6-8685-11e9-927d-00bb609c48fc/piksi_multi_base_station-2*.log
[piksi_multi_base_station-2] restarting process
process[piksi_multi_base_station-2]: started with pid [15492]
[INFO] [1559624230.130236]: /piksi_multi_base_station start
Traceback (most recent call last):
  File "/home/ocp-ais/.local/bin/pip", line 7, in <module>
    from pip._internal import main
ImportError: No module named _internal
[FATAL] [1559624230.562861]: No SBP library found. Please install it by using script in 'install' folder.
Traceback (most recent call last):
  File "/home/ocp-ais/swift_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/bin/piksi_multi", line 6, in <module>
    piksi_multi = PiksiMulti()
  File "/home/ocp-ais/swift_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/src/piksi_multi_rtk_ros/piksi_multi.py", line 89, in __init__
    rospy.loginfo("libsbp version currently used: " + installed_sbp_version)
TypeError: cannot concatenate 'str' and 'int' objects
[piksi_multi_base_station-2] process has died [pid 15492, exit code 1, cmd /home/ocp-ais/swift_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/bin/piksi_multi __name:=piksi_multi_base_station __log:=/home/ocp-ais/.ros/log/33abc9f6-8685-11e9-927d-00bb609c48fc/piksi_multi_base_station-2.log].
log file: /home/ocp-ais/.ros/log/33abc9f6-8685-11e9-927d-00bb609c48fc/piksi_multi_base_station-2*.log
[piksi_multi_base_station-2] restarting process
process[piksi_multi_base_station-2]: started with pid [15517]
[INFO] [1559624231.858224]: /piksi_multi_base_station start
Traceback (most recent call last):
  File "/home/ocp-ais/.local/bin/pip", line 7, in <module>
    from pip._internal import main
ImportError: No module named _internal
[FATAL] [1559624232.295760]: No SBP library found. Please install it by using script in 'install' folder.
Traceback (most recent call last):
  File "/home/ocp-ais/swift_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/bin/piksi_multi", line 6, in <module>
    piksi_multi = PiksiMulti()
  File "/home/ocp-ais/swift_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/src/piksi_multi_rtk_ros/piksi_multi.py", line 89, in __init__
    rospy.loginfo("libsbp version currently used: " + installed_sbp_version)
TypeError: cannot concatenate 'str' and 'int' objects
[piksi_multi_base_station-2] process has died [pid 15517, exit code 1, cmd /home/ocp-ais/swift_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/bin/piksi_multi __name:=piksi_multi_base_station __log:=/home/ocp-ais/.ros/log/33abc9f6-8685-11e9-927d-00bb609c48fc/piksi_multi_base_station-2.log].
log file: /home/ocp-ais/.ros/log/33abc9f6-8685-11e9-927d-00bb609c48fc/piksi_multi_base_station-2*.log
[piksi_multi_base_station-2] restarting process
process[piksi_multi_base_station-2]: started with pid [15542]
[INFO] [1559624233.576546]: /piksi_multi_base_station start
Traceback (most recent call last):
  File "/home/ocp-ais/.local/bin/pip", line 7, in <module>
    from pip._internal import main
ImportError: No module named _internal
[FATAL] [1559624234.014116]: No SBP library found. Please install it by using script in 'install' folder.
Traceback (most recent call last):
  File "/home/ocp-ais/swift_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/bin/piksi_multi", line 6, in <module>
    piksi_multi = PiksiMulti()
  File "/home/ocp-ais/swift_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/src/piksi_multi_rtk_ros/piksi_multi.py", line 89, in __init__
    rospy.loginfo("libsbp version currently used: " + installed_sbp_version)
TypeError: cannot concatenate 'str' and 'int' objects
[piksi_multi_base_station-2] process has died [pid 15542, exit code 1, cmd /home/ocp-ais/swift_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/bin/piksi_multi __name:=piksi_multi_base_station __log:=/home/ocp-ais/.ros/log/33abc9f6-8685-11e9-927d-00bb609c48fc/piksi_multi_base_station-2.log].
log file: /home/ocp-ais/.ros/log/33abc9f6-8685-11e9-927d-00bb609c48fc/piksi_multi_base_station-2*.log
[piksi_multi_base_station-2] restarting process
process[piksi_multi_base_station-2]: started with pid [15567]
^C[piksi_multi_base_station-2] killing on exit
Traceback (most recent call last):
  File "/home/ocp-ais/swift_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/bin/piksi_multi", line 6, in <module>
    piksi_multi = PiksiMulti()
  File "/home/ocp-ais/swift_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/src/piksi_multi_rtk_ros/piksi_multi.py", line 79, in __init__
    rospy.init_node('piksi')
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 323, in init_node
    raise rospy.exceptions.ROSInitException("init_node interrupted before it could complete")
rospy.exceptions.ROSInitException: init_node interrupted before it could complete
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
[UsefulLogs_files.zip](https://github.com/ethz-asl/ethz_piksi_ros/files/3263779/UsefulLogs_files.zip)

ebimor, 2019-08-09:
Seems like your pip is broken. This command solved it on my machine

sudo apt remove python-pip

rikba, 2019-10-03:
FYI we massively simplified the [installation script](https://github.com/ethz-asl/ethz_piksi_ros/blob/v1.11.0/piksi_multi_rtk_ros/install/install_piksi_multi.sh) in the newest version. You may want to have another try.

closed 2019-10-03 by rikba