ethz-asl/ethz_piksi_ros #90: GPS-UTC time offset

hitimo, 2019-08-24:
Shouldn't 
https://github.com/ethz-asl/ethz_piksi_ros/blob/master/utils/piksi_rtklib_postp/loadGPS.m#L2
be 18?
https://endruntechnologies.com/support/leap-seconds

michaelpantic, 2019-08-27:
Hm, so as per https://gssc.esa.int/navipedia/index.php/Transformations_between_Time_Systems, the relations are as following:

TAI (International Atomic time) is per definition 19 s ahead of GPST, so GPST = TAI  - 19

Currently, TAI is ahead of UTC by 37 Secs, ,so TAI = UTC + 37

so, GPST = UTC+ 37- 19 = UTC + 18, so it should be 18 then.

@rikba I just checked in the piksi driver, there it should be handled correctly. Given that pythons' leapseconds module works as it should ;-)
(https://github.com/ethz-asl/ethz_piksi_ros/blob/cc0a2ccba51f404b3e9ab097ab3bedb658b19427/piksi_multi_rtk_ros/src/piksi_multi_rtk_ros/piksi_multi.py#L688)

hitimo, 2019-08-28:
@michaelpantic I see.. thanks!

closed 2019-08-28 by hitimo