ethz-asl/ethz_piksi_ros #95: Package piksi_multi_rtk_ros can not be found

GCpiksi, 2019-09-19:
I am using Ubuntu 16.04 and ROS-kinetic. I have followed the installation procedure (install/install_piksi_multi.sh). The script completes without errors.

Now when I try to run the example:

 `roslaunch piksi_multi_rtk_ros piksi_multi_base_station.launch`

I get the following error:

`[piksi_multi_base_station.launch] is neither a launch file in package [piksi_multi_rtk_ros] nor is [piksi_multi_rtk_ros] a launch file name`

I have no idea how I can make roslaunch find the package as there is no setup.bash. I am confused.

rikba, 2019-09-19:
Additionally you'll need to build the catkin package, i.e.,

```
cd ~/catkin_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros
catkin build --this
```
Then you can source your workspace, i.e.,

```
source ~/catkin_ws/devel/setup.bash
``` 

and in that terminal you should be able to find the launch file then.

GCpiksi, 2019-09-19:
Thank you! I was convinced these were Python nodes and I have zero experience with those. Perhaps this could be added to the README.md for the less experienced users like myself.

In my ROS version `catkin build --this` did nothing. I built the whole workspace with `catkin_make -j1` to make it only use one thread. It took a while but it worked fine. All the launch files work now.

closed 2019-09-19 by GCpiksi