ethz-asl/ethz_piksi_ros #98: Code profiling

rikba, 2019-09-25:
The Piksi multi ROS driver occupies too much CPU. We need to profile the code and either fix the leak or implement a minimum cpp driver for sensitive tasks. @stlucas44

rikba, 2019-09-26:
Currently it uses up about 10% of a i7-7820HQ core.

[image]

michaelpantic, 2019-09-26:
Tipps to profile:

To install:

sudo pip install yappi
sudo apt-get install kcachegrind

To profile rosnodes, use a launch-prefix inside the roslaunch file:
launch-prefix="yappi -b -f callgrind -o /tmp/yappi.out"

(where /tmp/yappi.out is the output of the profiler)

To then examine the profiler files, use cachegrind:
kcachegrind /tmp/yappi.out

See also https://kcachegrind.github.io/html/Home.html

michaelpantic, 2019-09-26:
One additional note:

Profilers sometimes don't like it if the program to be profiled is just killed / ctrl-c' ed like we do it with rosnodes. To gracefully exit, you can also add an "Exit Service", thats just a rosservice that shuts down the node once called.

stlucas44, 2019-10-04:
The profiling is performed in a first iteration by adding the proposed launch-prefix:
launch-prefix= "yappi -b -f callgrind  -o /home/lucas/tmp/test.out"

The results were difficult to interpret because of confusing visualization. One assumes that the framer as well as the handler, which is operating the framer, are implemented in a slow manner. 

[image]

rikba, 2019-11-28:
CPP driver performs way better and to be used in future.

closed 2019-11-28 by rikba