ethz-asl/ethz_piksi_ros #102: piksi_multi_cpp development

rikba, 2019-10-08:
In https://github.com/ethz-asl/ethz_piksi_ros/pull/101 we introduced a first shot at a cpp ROS driver development. Several improvements and additions are necessary to make it usable.

- [ ] Implement serial devices
- [ ] Implement ethernet devices
- [ ] Implement `readSettings` methods
- [ ] Identify Piksi Multi, e.g., base, reference receiver, attitude receiver
- [ ] Expose all necessary SBP messages
- [ ] Visualize certain messages
- [ ] Multithreaded reading
- [ ] UML documentation
- [ ] Launch files
- [ ] Testing
- [ ] ROS publisher architecture
- [ ] Geo frame conversion

michaelpantic, 2019-10-10:
I'll implement serial devices and publishing of corrections some time next week!

rikba, 2019-10-10:
[image]

michaelpantic, 2019-10-10:
Btw, that was the uml tool that @clanegge  used: http://plantuml.com/

rikba, 2019-10-23:
Potential structure of SettingsIo/Receiver classes @michaelpantic 

[image]

closed 2022-05-06 by rikba