ethz-asl/ethz_piksi_ros #115: libsbp_catkin error

onatkeser, 2019-10-16:
Hi I receive this error while doing catkin_make:
By not providing "Findlibsbp_catkin.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "libsbp_catkin", but CMake did not find one.

  Could not find a package configuration file provided by "libsbp_catkin"
  with any of the following names:

    libsbp_catkinConfig.cmake
    libsbp_catkin-config.cmake

  Add the installation prefix of "libsbp_catkin" to CMAKE_PREFIX_PATH or set
  "libsbp_catkin_DIR" to a directory containing one of the above files.  If
  "libsbp_catkin" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  ethz_piksi_ros/piksi_multi_cpp/CMakeLists.txt:8 (catkin_simple)

-- Configuring incomplete, errors occurred!

Could you please help me about that?

onatkeser, 2019-10-16:
And this is my output after executing the install file(I guess this causes the issue above):

/ethz_piksi_ros/piksi_multi_rtk_ros$ source install/install_piksi_multi.sh 
 
Installing SBP library for Piksi Multi.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python-pip is already the newest version (8.1.1-2ubuntu0.4).
The following packages were automatically installed and are no longer required:
  clamav clamav-base clamav-freshclam fltk1.3-doc gir1.2-gtk-2.0 icoutils jq
  libatk1.0-dev libavdevice-dev libavfilter-dev libavresample-dev
  libcairo-script-interpreter2 libcairo2-dev libclamav7 libfltk-cairo1.3
  libfltk-forms1.3 libfltk-gl1.3 libfltk-images1.3 libfltk1.3 libfltk1.3-dev
  libgdk-pixbuf2.0-dev libgtk2.0-dev libharfbuzz-dev libharfbuzz-gobject0
  libignition-cmake-dev libignition-common libignition-common-dev
  libignition-fuel-tools-dev libignition-fuel-tools1-1
  libignition-fuel-tools1-dev libignition-math4 libignition-math4-dev
  libignition-msgs libignition-msgs-dev libignition-transport4
  libignition-transport4-dev libllvm3.6v5 libonig2 libpango1.0-dev
  libpixman-1-dev libpostproc-dev libqwt-headers libqwt-qt5-6 libqwt-qt5-dev
  libruby2.3 libtext-csv-perl libtext-csv-xs-perl libxcb-shm0-dev
  libxcomposite-dev libxcursor-dev libxinerama-dev libxrandr-dev libyaml-dev
  libzip-dev libzip4 libzmq3-dev par2 protobuf-compiler python-avahi
  python-gdbm python-opengl python-pyqt5.qtopengl python-pyqt5.qtwebkit
  python3-dateutil python3-pluggy python3-py python3-virtualenv rake
  ros-kinetic-camera-calibration-parsers ros-kinetic-camera-info-manager
  ros-kinetic-geographic-msgs ros-kinetic-gl-dependency
  ros-kinetic-interactive-marker-twist-server
  ros-kinetic-master-discovery-fkie ros-kinetic-master-sync-fkie
  ros-kinetic-multimaster-msgs ros-kinetic-multimaster-msgs-fkie
  ros-kinetic-pointcloud-to-laserscan ros-kinetic-polled-camera
  ros-kinetic-robot-localization ros-kinetic-teleop-twist-joy
  ros-kinetic-tf2-geometry-msgs ros-kinetic-tf2-sensor-msgs
  ros-kinetic-twist-mux ros-kinetic-twist-mux-msgs ros-kinetic-uuid-msgs
  ros-kinetic-velodyne-description ros-kinetic-webkit-dependency ruby
  ruby-did-you-mean ruby-minitest ruby-net-telnet ruby-power-assert
  ruby-test-unit ruby2.3 rubygems-integration snapd-login-service tox
  ttf-liberation virtualenv x11proto-composite-dev x11proto-randr-dev
  x11proto-xinerama-dev
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 604 not upgraded.
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    from pip import main
ImportError: cannot import name main
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    from pip import main
ImportError: cannot import name main
SBP Library Installed
Installing ROS node requirements.
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    from pip import main
ImportError: cannot import name main
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    from pip import main
ImportError: cannot import name main
Traceback (most recent call last):
  File "/usr/bin/pip", line 9, in <module>
    from pip import main
ImportError: cannot import name main
User emreonat is already included in 'dialout' group.
 
Installation completed.

rikba, 2019-10-17:
Hi

Please use a stable release, e.g., https://github.com/ethz-asl/ethz_piksi_ros/tree/v1.11.0. The current master is under development.

milano-technical-intern, 2019-11-02:
@rikba I'm using the v.1.11.0 version and I'm getting the same error. Even I need to install catkin_simple while previously was not needed.