ethz-asl/ethz_piksi_ros #119: RTK mode not working

luissantos27, 2019-10-22:
The driver is providing data to every topic except the ones where RTK mode is needed. 
There no GNSS signal fail because the system works with the swift console or even with the discontinued driver "swiftnav_ros". Is there any kind of incompatibility with the new firmware or am I using faulty parameters? 

My firmware version is: 2.3.19

`roslaunch piksi_multi_rtk_ros piksi_multi_rover.launch`

`<?xml version="1.0"?>
<launch>
  
  
  <arg name="interface"                   default="serial"/>
  
  <arg name="baud_rate"                   default="115200"/>
  
  <arg name="tcp_addr"                    default="192.168.0.222"/>
  
  <arg name="base_station_ip"             default="192.168.0.1"/>
  
  <arg name="load_enu_origin_from_file"   default="false"/>
  
  <arg name="enu_origin_file"             default="$(find piksi_multai_rtk_ros)/cfg/enu_origin_example.yaml"/>

  
  <node pkg="piksi_multi_rtk_ros" type="piksi_multi" name="piksi" output="screen" respawn="true">
    
    <rosparam file="$(find piksi_multi_rtk_ros)/cfg/piksi_multi_driver_settings.yaml"/>

    
    <rosparam file="$(arg enu_origin_file)"   if="$(arg load_enu_origin_from_file)"/>

    
    <param name="interface"                               value="$(arg interface)"/>
    <param name="baud_rate"                               value="$(arg baud_rate)"/>
    <param name="tcp_addr"                                value="$(arg tcp_addr)"/>
    <param name="base_station_ip_for_latency_estimation"  value="$(arg base_station_ip)"/>
    
    <param name="base_station_mode"                       value="false"/>
  </node>

</launch>`

PIKISI_MULTI_DRIVER_SETTINGS.YAML

`# General Settings

# interface: specifies the interface over which to connect.
# supported values are 'serial' and 'tcp'
# if this value is not valid, the driver will default to serial communication.
interface: 'serial'

# set this to true if the Swift device is acting as RTK Base Station.
base_station_mode: false

# if true, output every topic read by this driver.
debug_mode: false

# if true the driver is verbose.
driver_verbose: true

# if true, raw imu and magnetometer are published
# (messages are available only if enabled with the swiftnav console).
publish_raw_imu_and_mag: false

# broadcast_addr: ip address to use for broadcasting corrections
# whenever possible, use network broadcast, not global ip broadcast
# use ifconfig to show network broadcast (or calculate it yourself....).
broadcast_addr: 192.168.0.255

# broadcast_port: port to use for broadcasting corrections.
broadcast_port: 26078

# base_station_ip_for_latency_estimation: IP of base station for
# latency estimation.
base_station_ip_for_latency_estimation: 192.168.0.1

# Coordinate frame of the GPS receiver with respect to the base link.
navsatfix_frame_id: 'base_link'

# Covariance Settings
# Variance to be published in the NavSatFix message (in diagonal part).
# Single Point Positioning (SPP).
var_spp: [25.0, 25.0, 64.0]
# Single Point Positioning (SPP) with SBAS.
var_spp_sbas: [1.0, 1.0, 1.0]
# RTK float mode.
var_rtk_float: [25.0, 25.0, 64.0]
# RTK fix mode.
var_rtk_fix: [0.0049, 0.0049, 0.01]

# Serial Settings

# serial_port: defines which serial device to use for communication
# ex '/dev/ttyUSB0'.
serial_port: '/dev/ttyUSB0'

# baud_rate: defines the rate for the serial connection. This driver uses
# 230400 as the default. Swift Navigation receivers ship with a
# default baud rate of 115200.
baud_rate: 115200

# TCP Settings

# tcp_addr: the IP address of the receiver. By default, Swift Navigation
# receivers have an IP address of 192.168.0.222.
tcp_addr: 192.168.0.222

# tcp_port: the port number of the TCP server. By default, Swift Navigation
# receivers publish SBP data on TCP port 55555.
tcp_port: 55555

# Time settings.
# use_gps_time: If true, it timestamps the position information with GPS time.
# Otherwise uses rospy.Time.now()
use_gps_time: false
`

ENU_ORIGIN_EXAMPLE.YAML

`latitude0_deg: 47.81
longitude0_deg: 8.51
altitude0: 565.7`

rikba, 2019-10-22:
Hi

Can you print the console output of the node?

rikba, 2019-10-22:
Also make sure you are using our most recent release https://github.com/ethz-asl/ethz_piksi_ros/tree/v1.11.0

ziliHarvey, 2019-10-25:
Try to enable debug mode. I met exactly the same problem and solved by doing this.

luissantos27, 2019-10-28:
@rikba I am using the most recent release (v 1.11.0) as you can check:

```
user@white:~/catkin_ws_h/src/ethz_piksi_ros$ git show
commit 2f3c0a864f0768734ecbf7ef36ef71fd9cc2ab7e
Merge: 11edddf db455cc
Author: Rik Baehnemann <brik@ethz.ch>
Date:   Thu Oct 3 19:05:43 2019 +0200

    Merge pull request #92 from ethz-asl/feature/sbp_v2.6.3

    Feature/sbp v2.6.3

```

Here is my console output:

```
user@white:~/catkin_ws_h/src/white_nav_conf/launch/hardware/driver_gps_piksi$ roslaunch run_driver_gps_piksi_new.launch 
... logging to /home/user/.ros/log/40e19814-f979-11e9-9f1e-7085c21cf97b/roslaunch-white-14689.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://white:46723/

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /white/piksi/altitude0: 186.0
 * /white/piksi/base_station_ip_for_latency_estimation: 192.168.0.1
 * /white/piksi/base_station_mode: False
 * /white/piksi/baud_rate: 115200
 * /white/piksi/broadcast_addr: 192.168.0.255
 * /white/piksi/broadcast_port: 26078
 * /white/piksi/debug_mode: False
 * /white/piksi/driver_verbose: True
 * /white/piksi/interface: serial
 * /white/piksi/latitude0_deg: 41.178946
 * /white/piksi/longitude0_deg: 8.595137
 * /white/piksi/navsatfix_frame_id: base_link
 * /white/piksi/publish_raw_imu_and_mag: False
 * /white/piksi/serial_port: /dev/gps_rover_rtk
 * /white/piksi/tcp_addr: 192.168.0.222
 * /white/piksi/tcp_port: 55555
 * /white/piksi/use_gps_time: False
 * /white/piksi/var_rtk_fix: [0.0049, 0.0049, ...
 * /white/piksi/var_rtk_float: [25.0, 25.0, 64.0]
 * /white/piksi/var_spp: [25.0, 25.0, 64.0]
 * /white/piksi/var_spp_sbas: [1.0, 1.0, 1.0]

NODES
  /white/
    piksi (piksi_multi_rtk_ros/piksi_multi)

ROS_MASTER_URI=http://localhost:11311

process[white/piksi-1]: started with pid [14706]
[INFO] [1572272395.376030]: /white/piksi start
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
[INFO] [1572272395.777406]: libsbp version currently used: 2.6.5
[INFO] [1572272395.783243]: Swift driver started in normal mode.
[INFO] [1572272395.801963]: Origin ENU frame set to: 41.178946, 8.595137, 186.00
[INFO] [1572272395.802416]: Origin ENU frame set by rosparam.
[INFO] [1572272395.838159]: Starting device in Rover Mode
Host Side Unhandled byte: 0xa4
^C[white/piksi-1] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done
```

luissantos27, 2019-10-28:
> Try to enable debug mode. I met exactly the same problem and solved by doing this.

I had already tried to enable debug mode, but the rtk topic still empty.

luissantos27, 2019-10-28:
@rikba 
Why do you use a different baud rate from the swiftnav default? You are using 230400 when the swiftnav works with 115200. Could this be an issue?

luissantos27, 2019-10-29:
I should have mentioned that I was using the FreeWave provided with the swiftnav multi kit ([here](https://www.swiftnav.com/store/piksi-multi-evaluation-kit/piksi-multi-evaluation-kit-2.4ghz)). 

I had to change the baud rate in both devices by connecting them to the swift console. The UART-1 should be 230400. 

Then, connect the base to a console and set the survey position to true. Click in auto survey and Save to device. It should be good to go after this!

rikba, 2019-11-04:
Hi @luissantos27 

> Why do you use a different baud rate from the swiftnav default?

We changed some of the baud rates for better data transfer. Especially, when time stamping the GPS data on arrival. But my recommendation would be to use the USB or Ethernet interface to obtain the data.

> I had to change the baud rate in both devices by connecting them to the swift console. The UART-1 should be 230400.

Can we close this issue?

luissantos27, 2019-11-05:
> Hi @luissantos27
>
>
> We changed some of the baud rates for better data transfer. Especially, when time stamping the GPS data on arrival. But my recommendation would be to use the USB or Ethernet [...]

Yes this issue is closed to me

closed 2019-11-05 by rikba