ethz-asl/ethz_piksi_ros #122: libsbp_catkin error

milano-technical-intern, 2019-11-02:
Hi,

I git cloned this [repository](https://github.com/ethz-asl/ethz_piksi_ros/tree/v1.11.0). After executed the command: catkin_make

I'm getting the following errors:

```
Traceback (most recent call last):
  File "/home/mgt/ros_wolverine/src/ethz_piksi_ros/libsbp_ros_msgs/bin/generator", line 5, in <module>
    import libsbp_ros_msgs.ros_renderer as renderer
  File "/home/mgt/ros_wolverine/src/ethz_piksi_ros/libsbp_ros_msgs/src/libsbp_ros_msgs/ros_renderer.py", line 4, in <module>
    **import jinja2**
ImportError: No module named jinja2
CMake Error at /home/mgt/ros_wolverine/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:38 (find_package):
  By not providing "Findlibsbp_catkin.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "libsbp_catkin", but CMake did not find one.
  **Could not find a package configuration file provided by "libsbp_catkin"
  with any of the following names:**

    libsbp_catkinConfig.cmake
    libsbp_catkin-config.cmake

  Add the installation prefix of "libsbp_catkin" to CMAKE_PREFIX_PATH or set
  "libsbp_catkin_DIR" to a directory containing one of the above files.  If
  "libsbp_catkin" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  ethz_piksi_ros/libsbp_ros_msgs/CMakeLists.txt:24 (catkin_simple)

-- Configuring incomplete, errors occurred!
See also "/home/mgt/ros_wolverine/build/CMakeFiles/CMakeOutput.log".
See also "/home/mgt/ros_wolverine/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```
Similar to this but the issue was closed.
#115

rikba, 2019-11-04:
Hi @milano-technical-intern

[v1.11.0](https://github.com/ethz-asl/ethz_piksi_ros/tree/v1.11.0) should not depend on libsbp_catkin or catkin_simple. Please make sure that you checkout the right branch.

```
git clone https://github.com/ethz-asl/ethz_piksi_ros.git
cd ethz_piksi_ros
git checkout v1.11.0
```

milano-technical-intern, 2019-11-05:
@rikba Thanks! That did the trick!

closed 2019-11-05 by milano-technical-intern

zlg9folira, 2019-12-27:
I have the same error, I am trying to build `libsbp_ros_msgs` and I have all other packages installed already. However, the workaround above didn't fix the issue. 

`could not find a package configuration file provided by "libsbp_catkin" with any of the following names:
libsbp_catkinconfig.cmake
libsbp_catkin-config.cmake
`

rikba, 2020-01-03:
Hi @zlg9folira 

if you checkout v1.11.0 there should be no dependency on libsbp_caktin and libsbp_ros_msgs should not exist. They are dependencies of a new C++ driver that only exist on the master branch. But if you want you can also build the master branch. It requires https://github.com/ethz-asl/libsbp_catkin, see also https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/install/dependencies.rosinstall