ethz-asl/ethz_piksi_ros #129: Module not found error - No module named 'piksi_multi'

james-yoo, 2019-11-18:
I am using Duro Inertial and would like to integrate ROS. I have installed piksi_multi driver with v1.10.0 stable release and launched pikisi_multi_rover.launch file. But following error is occurred with module missing message. Does anybody know how to solve this issue?
----------------------------------------------------------------------------------------------------------------
started core service [/rosout]
process[piksi-2]: started with pid [30833]
Traceback (most recent call last):
  File "/home/nvidia/catkin_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/bin/piksi_multi", line 3, in <module>
    from  piksi_multi_rtk_ros import PiksiMulti
  File "/home/nvidia/catkin_ws/devel/lib/python2.7/dist-packages/piksi_multi_rtk_ros/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
**ModuleNotFoundError: No module named 'piksi_multi'**
[piksi-2] process has died [pid 30833, exit code 1, cmd /home/nvidia/catkin_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/bin/piksi_multi __name:=piksi __log:=/home/nvidia/.ros/log/9f47bec8-0a5a-11ea-aa47-00044bcbac1b/piksi-2.log].
log file: /home/nvidia/.ros/log/9f47bec8-0a5a-11ea-aa47-00044bcbac1b/piksi-2*.log

skyetheguy, 2020-10-27:
Kind of late now but I'm having the same problem.. Did you ever find a solution?

ColtenBecker, 2021-03-29:
@skyetheguy @james-yoo Did either of you find a fix for this? I am having the same error.

Jonny-air, 2021-05-27:
Hi, did any of you solve this? I am having the same issue but with v1.11.0. I am also using ROS noetic.

Jonny-air, 2021-05-27:
I was able to solve it in my case by adding a "." to the import statements in the __init__.py file:

from . piksi_multi import PiksiMulti
from . geodetic_survey import GeodeticSurvey

This was necessary since in python3 all imports are absolute and not relative by default. Not sure if this is the same issue that you are having, since I am using python3, but may help someone in the future.

broccoli-smuggler, 2021-07-22:
I've made a fork with changes that runs for me with python3: should also resolve your issues:
https://github.com/broccoli-smuggler/ethz_piksi_ros/tree/feature/ubunutu_20_python3

broccoli-smuggler, 2021-07-22:
It's based off the v1.11.0.

chouaibharik, 2021-08-11:
Hi,

@broccoli-smuggler I tried the new fork for python3 and ubuntu 20.04 ( I do have Noetic installed), but it didn't work.

Here is what i get when i launch the piksi_multi_rover.launch:

PARAMETERS
 * /piksi/base_station_ip_for_latency_estimation: 192.168.0.1
 * /piksi/base_station_mode: False
 * /piksi/baud_rate: 230400
 * /piksi/broadcast_addr: 192.168.0.255
 * /piksi/broadcast_port: 26078
 * /piksi/debug_mode: False
 * /piksi/driver_verbose: True
 * /piksi/interface: serial
 * /piksi/navsatfix_frame_id: base_link
 * /piksi/publish_raw_imu_and_mag: False
 * /piksi/serial_port: /dev/ttyUSB0
 * /piksi/tcp_addr: 192.168.0.222
 * /piksi/tcp_port: 55555
 * /piksi/use_gps_time: False
 * /piksi/var_rtk_fix: [0.0049, 0.0049, ...
 * /piksi/var_rtk_float: [25.0, 25.0, 64.0]
 * /piksi/var_spp: [25.0, 25.0, 64.0]
 * /piksi/var_spp_sbas: [1.0, 1.0, 1.0]
 * /rosdistro: noetic
 * /rosversion: 1.15.11

NODES
  /
    piksi (piksi_multi_rtk_ros/piksi_multi)

auto-starting new master
process[master]: started with pid [148548]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to bab2d432-fa85-11eb-be8e-8772cdcf4195
process[rosout-1]: started with pid [148561]
started core service [/rosout]
process[piksi-2]: started with pid [148568]
generating LLVM code for 'parse_jit_py38' into /tmp/pycc-build-parse_jit_py38-avwde0b5/parse_jit_py38.cpython-38-x86_64-linux-gnu.o
C compiler: x86_64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC

creating /tmp/pycc-build-parse_jit_py38-avwde0b5/home
creating /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch
creating /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local
creating /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local/lib
creating /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local/lib/python3.8
creating /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local/lib/python3.8/site-packages
creating /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local/lib/python3.8/site-packages/numba
creating /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local/lib/python3.8/site-packages/numba/pycc
creating /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local/lib/python3.8/site-packages/numba/cext
creating /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local/lib/python3.8/site-packages/numba/core
creating /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local/lib/python3.8/site-packages/numba/core/runtime
compile options: '-DPYCC_MODULE_NAME=parse_jit_py38 -DPYCC_USE_NRT=1 -I/usr/include/python3.8 -I/home/ehch/.local/lib/python3.8/site-packages/numpy/core/include -c'
x86_64-linux-gnu-gcc: /home/ehch/.local/lib/python3.8/site-packages/numba/cext/utils.c
x86_64-linux-gnu-gcc: /home/ehch/.local/lib/python3.8/site-packages/numba/cext/listobject.c
x86_64-linux-gnu-gcc: /home/ehch/.local/lib/python3.8/site-packages/numba/pycc/modulemixin.c
x86_64-linux-gnu-gcc: /home/ehch/.local/lib/python3.8/site-packages/numba/cext/dictobject.c
x86_64-linux-gnu-gcc: /home/ehch/.local/lib/python3.8/site-packages/numba/pycc/../core/runtime/nrt.c
x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /tmp/pycc-build-parse_jit_py38-avwde0b5/parse_jit_py38.cpython-38-x86_64-linux-gnu.o /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local/lib/python3.8/site-packages/numba/pycc/modulemixin.o /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local/lib/python3.8/site-packages/numba/cext/utils.o /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local/lib/python3.8/site-packages/numba/cext/listobject.o /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local/lib/python3.8/site-packages/numba/cext/dictobject.o /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local/lib/python3.8/site-packages/numba/core/runtime/nrt.o -L/home/ehch/.local/lib/python3.8/site-packages/numpy/core/lib -lnpymath -lm -o /usr/local/lib/python3.8/dist-packages/sbp/jit/parse_jit_py38.cpython-38-x86_64-linux-gnu.so
Traceback (most recent call last):
  File "/usr/lib/python3.8/distutils/unixccompiler.py", line 215, in link
    self.spawn(linker + ld_args)
  File "/home/ehch/.local/lib/python3.8/site-packages/numpy/distutils/ccompiler.py", line 88, in <lambda>
    m = lambda self, *args, **kw: func(self, *args, **kw)
  File "/home/ehch/.local/lib/python3.8/site-packages/numpy/distutils/ccompiler.py", line 171, in CCompiler_spawn
    raise DistutilsExecError('Command "%s" failed with exit status %d%s' %
distutils.errors.DistutilsExecError: Command "x86_64-linux-gnu-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -Wl,-Bsymbolic-functions -Wl,-z,relro -g -fwrapv -O2 -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 /tmp/pycc-build-parse_jit_py38-avwde0b5/parse_jit_py38.cpython-38-x86_64-linux-gnu.o /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local/lib/python3.8/site-packages/numba/pycc/modulemixin.o /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local/lib/python3.8/site-packages/numba/cext/utils.o /tmp/pycc-build-parse_jit_py38-avwde0b5/home/ehch/.local/lib/python3.8/site-packag [... cut ...] avwde0b5/home/ehch/.local/lib/python3.8/site-packages/numba/core/runtime/nrt.o -L/home/ehch/.local/lib/python3.8/site-packages/numpy/core/lib -lnpymath -lm -o /usr/local/lib/python3.8/dist-packages/sbp/jit/parse_jit_py38.cpython-38-x86_64-linux-gnu.so" failed with exit status 1
[piksi-2] process has died [pid 148568, exit code 1, cmd /home/ehch/workspace/catkin_ws/devel/lib/piksi_multi_rtk_ros/piksi_multi __name:=piksi __log:=/home/ehch/.ros/log/bab2d432-fa85-11eb-be8e-8772cdcf4195/piksi-2.log].

chouaibharik, 2021-08-12:
@broccoli-smuggler I confirm that it works on Ubuntu 20.04 with Ros Noetic installed. Thanks a lot!

tomaszkapron, 2022-04-08:
If someone else is still having @ChouBot issue with @broccoli-smuggler fork, it helped to update sbd package to version 3.4.11.

rikba, 2022-05-06:
Thanks @broccoli-smuggler for your fork!

If you still want to use this repository, please use [piksi_multi_cpp](https://github.com/ethz-asl/ethz_piksi_ros/tree/master/piksi_multi_cpp) with ROS noetic and Piksi Multi Firmware 2.4.16. The Python driver is not maintained and outdated.

closed 2022-05-06 by rikba