ethz-asl/ethz_piksi_ros #135: please make UdpHelpers.py python3 compliant

jbreiden2, 2019-11-22:
```diff
--- UdpHelpers.py	2019-05-10 09:07:25.000000000 -0700
+++ UdpHelpers.py	2019-11-21 11:07:18.000000000 -0800
@@ -37,7 +37,7 @@
         try:
             #self.handle.connect((host, port))
             self.handle.bind(("", port))
-        except socket.error:
+        except socket.error, msg:
             pass
         super(UDPDriver, self).__init__(self.handle)
         self._write_lock = threading.Lock()
@@ -47,11 +47,11 @@
             try:
                 data, addr = self.handle.recvfrom(4096)
                 if not data:
-                    print("PIKSI UDP ERROR - no data from " + str(addr))
+                    print "PIKSI UDP ERROR - no data from " + str(addr)
                 for d in data:
                     self.buf.append(d)
-            except socket.error as socket_error:
-                print("PIKSI UDP ERROR " + socket_error)
+            except socket.error, msg:
+                print "PIKSI UDP ERROR " + str(msg)
 
         res = ''.join([self.buf.popleft() for i in xrange(min(size, len(self.buf)))])
         return res
@@ -71,7 +71,7 @@
         try:
             self._write_lock.acquire()
             self.handle.sendall(s)
-        except socket.error:
+        except socket.error, msg:
             raise IOError
         finally:
             self._write_lock.release()
```

jbreiden2, 2019-11-22:
Sorry, that was the wrong patch. Here is the right one.

```diff
--- UdpHelpers.py	2019-11-21 11:07:18.000000000 -0800
+++ UdpHelpers.py	2019-11-21 13:29:13.000000000 -0800
@@ -22,6 +22,7 @@
 
 # Adapted from https://github.com/uscresl/piksi_ros/blob/master/src/piksi_driver.py
 
+from __future__ import print_function
 from socket import *
 from sbp.client.drivers.base_driver import BaseDriver
 from sbp.client import Handler, Framer
@@ -37,7 +38,7 @@
         try:
             #self.handle.connect((host, port))
             self.handle.bind(("", port))
-        except socket.error, msg:
+        except socket.error as msg:
             pass
         super(UDPDriver, self).__init__(self.handle)
         self._write_lock = threading.Lock()
@@ -47,11 +48,11 @@
             try:
                 data, addr = self.handle.recvfrom(4096)
                 if not data:
-                    print "PIKSI UDP ERROR - no data from " + str(addr)
+                    print("PIKSI UDP ERROR - no data from " + str(addr))
                 for d in data:
                     self.buf.append(d)
-            except socket.error, msg:
-                print "PIKSI UDP ERROR " + str(msg)
+            except socket.error as msg:
+                print("PIKSI UDP ERROR " + str(msg))
 
         res = ''.join([self.buf.popleft() for i in xrange(min(size, len(self.buf)))])
         return res
@@ -71,7 +72,7 @@
         try:
             self._write_lock.acquire()
             self.handle.sendall(s)
-        except socket.error, msg:
+        except socket.error as msg:
             raise IOError
         finally:
             self._write_lock.release()
```

rikba, 2022-05-06:
Please use [piksi_multi_cpp](https://github.com/ethz-asl/ethz_piksi_ros/tree/master/piksi_multi_cpp) with ROS noetic and Piksi Multi Firmware 2.4.16. The Python driver is not maintained and outdated. If you still want to use the Python driver with Python3 we encourage a fork.

closed 2022-05-06 by rikba

rikba, 2022-05-06:
One fork by broccoli-smuggler is described here: https://github.com/ethz-asl/ethz_piksi_ros/issues/129

https://github.com/broccoli-smuggler/ethz_piksi_ros/tree/feature/ubunutu_20_python3