ethz-asl/ethz_piksi_ros #139: please add flag for skylark support

jbreiden2, 2019-11-22:
The Skylark configuration generates rtk_float exclusively. Adds a flag to make it work.

```diff
--- piksi_multi.py	2019-11-21 11:07:18.000000000 -0800
+++ piksi_multi.py	2019-11-21 13:29:13.000000000 -0800
@@ -171,6 +171,9 @@
         self.enu_frame_id = rospy.get_param('~enu_frame_id', 'enu')
         self.transform_child_frame_id = rospy.get_param('~transform_child_frame_id', 'gps_receiver')
 
+        self.skylark = rospy.get_param('~skylark', False)
+
         if rospy.has_param('~latitude0_deg') and rospy.has_param('~longitude0_deg') and rospy.has_param(
                 '~altitude0'):
             latitude0 = rospy.get_param('~latitude0_deg')
@@ -777,7 +780,8 @@
             return
         # RTK messages.
         elif fix_mode == PosLlhMulti.FIX_MODE_FLOAT_RTK:
-            if self.origin_enu_set:
+            if self.origin_enu_set or self.skylark:
                 self.publish_rtk_float(msg.lat, msg.lon, msg.height, stamp, self.var_rtk_float)
             else:
                 rospy.logwarn_throttle(5,
```

sicongw, 2020-04-29:
Pull request https://github.com/ethz-asl/ethz_piksi_ros/pull/168 proposes to automatically initialize ENU origin upon receiving the first RTK Float packet, which will allow Skylark mode to work without adding additional flags.

Suggest closing this issue if #168 gets approved.

rikba, 2022-05-06:
Unfortunately, we do not have Skylark services and can test this.

closed 2022-05-06 by rikba