ethz-asl/ethz_piksi_ros #152: CMake error at ethz_piksi_ros/libsbp_ros_msgs/CMakeLists.txt:9

deepika164, 2020-01-17:
Hi, I am trying to launch the piksi multi base station and rover but process dies continuously. The baud rates in the cfg and launch files have been corrected but i get an error while building the messages.
I am following the steps from https://support.swiftnav.com/customer/en/portal/articles/2924342-using-ros-with-swift-navigation-gnss-devices   but it does not seem to work. 
This step in the above link : catkin_make --pkg piksi_rtk_msgs  gives an error saying invoking "make cmake_check_build_system"failed.

Please help 
Thanks

rikba, 2020-01-19:
Hi @deepika164 
can you post the full

deepika164, 2020-01-20:
yes, this is what it says

CMake Error at ethz_piksi_ros/libsbp_ros_msgs/CMakeLists.txt:9 (find package): By not providing "Findcatkin_simple.cmake" in CMAKE_MODULE_PATH this project has asked CMake to find a package configuration file provided by "catkin_simple", but CMake did not find one. 
Could not find a package configuration file provided by "catkin_simple" with any of the following names:
catkin_simpleConfig.cmake
catkin_simple-config.cmake
Makefile:318: recipe for target 'cmake_check_build_system' failed'
invoking "make cmake_check_build_system"failed

rikba, 2020-01-20:
The problem are missing dependencies of the latest C++ driver.
If you want to use our documented python driver, please checkout the latest release https://github.com/ethz-asl/ethz_piksi_ros/tree/v1.11.0. 
If you want to use our latest C++ driver, you need to install the following dependencies into your catkin workspace. https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/install/dependencies_https.rosinstall

deepika164, 2020-01-22:
I tried installing the C++ driver using sudo apt install  https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/install/dependencies_https.rosinstall but i get error saying "unable to locate package  https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/install/"
Couldn't find any package by glob " https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/install"
Couldn't find any package by regex " https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/install/dependencies_https.rosinstall"

Am I using the right commands?

deepika164, 2020-01-23:
Sorry, I found the error to my previous post, however i am faced with another error.
I git cloned each of the packages for C++ driver from the link sent to me and now i get error with glog_catkin.
CMake Error at /home/administrator/Documents/swift_ros_ws/devel/share/glog_catkin/cmake/glog_catkinConfig.cmake:150 (message): 
Project 'eigen_checks' tried to find library 'glog'. The library is neither a target nor built/installed properly. Did you compile project 'glog_catkin'? Did you find_package() it before the subdirectory containing its code is included?
Makefile:318: recipe for target 'cmake_check_build_system' failed'
invoking "make cmake_check_build_system"failed

I git cloned glog_catkin from here https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/install/dependencies_https.rosinstall so I don't know why i am getting this error.

Please help 
Thanks

pmaini, 2020-07-20:
I was having the same issue for quite some time. What worked for me is changing the order of compilation. The error above tells eigen_checks can't find glog_catkin. This could occur even if glog_catkin is already in your catkin workspace because the packages are compiled alphabetically. The best order is to start with clean workspace, initialize it. Then add glog_catkin, catkin_make, then add eigen_checks, catkin_make, then add catkin_simple, catkin_make, eigen_simple, catkin_make, then rest of dependencies and then add this package and compile. I hope this helps. Also I did not need the cpp package so i removed, cpp pkg and interface pkg. These two packages have a dependency on libgpiod. I could not get the ros system to correctly detect it after spending hours of effort. I didnt need them so just removed them and compiled and it works. I hope this helps you. I struggled quite a bit and have solved it so shoot any more questions that you might have, I will try to help as best as I can.

Best,
Parikshit

rikba, 2022-05-06:
Please use [piksi_multi_cpp](https://github.com/ethz-asl/ethz_piksi_ros/tree/master/piksi_multi_cpp) with ROS noetic and Piksi Multi Firmware 2.4.16. The Python driver is not maintained and outdated. If you still want to use the Python driver checkout release [v1.10.0](https://github.com/ethz-asl/ethz_piksi_ros/tree/v1.11.0).

closed 2022-05-06 by rikba