ethz-asl/ethz_piksi_ros #157: residual ns alway 0 with piksi_multi_rtk_gps

SamMuenchow, 2020-02-13:
Hi, For some reason I can't figure out are the residual nano seconds in the GPS_time message are always 0. All other data seems to be correct. I am using ros melodic , SBP Lib 2.6.5 and Firmware 2.3.19. 
Is there an issue with the residual ns?

Thanks for your help.

SamMuenchow, 2020-03-17:
I've looked deeper in the code and found out that the ns_residual is already zero when it is received by the SBP Lib. Then I asked Swiftnav directly if this is a bug and they told me that this behaivior is intended. The device sends its information at the moment when the TOW second change and so the ns_residual should be zero.

However, the accuracy of this time information is by far not as good as the accuracy of the PPS signal.

closed 2020-03-17 by SamMuenchow

rikba, 2020-03-17:
Hi Sam

> However, the accuracy of this time information is by far not as good as the accuracy of the PPS signal.
The PPS signal is just as accurate as the time stamp of the GNSS solution provided by Swift. You could consider synchronizing your system with the PPS signal to fuse the information with other sensor data. https://github.com/ethz-asl/ethz_piksi_ros/wiki/Piksi-PPS-Sync

SamMuenchow, 2020-03-17:
Hi Rik

Yeah this is what I am now planning to do.

The problem with the time stamp is, that the transmission via ethernet and the processing time on the development board and the computer, takes quite a while and takes a different time to be done. My tests have showed me a delay up to 80ms.

Next step is to use the PPS Signal with a GPIO Port to reduce this delay.

rikba, 2020-03-17:
Sounds good to me 👍