ethz-asl/ethz_piksi_ros #170: Intertial Imu and Mag rates slow in ROS output

chriswsuarez, 2020-05-15:
Not sure if this an issue, by design, or if I'm missing some setting somewhere but I'm seeing that the /imu_raw and /mag_raw output rates in ROS are scaled back by a factor of 25 and 50 respectively.

IMU set to 200 Hz comes out as 4 Hz in the rostopic.
MAG set to 25 Hz comes out as 0.5 Hz in the rostopic.

These data rates in ROS are unusable as a primary IMU for my mobile robot.

[image]

[image]

pietrocolombo, 2020-07-27:
i have the same issue

EpicMinister, 2020-07-30:
same issue too

rikba, 2020-07-30:
Hi, can you give some information on what version of the driver you are using?

pietrocolombo, 2020-07-31:
ethz_piksi_ros:
v1.11.0
SBP Lib:
v2.6.5
Firmware:
v2.3.19

EpicMinister, 2020-08-13:
I had the same issue, the support from Swift Nav helped me.

This is no ROS or driver problem. In the settings of your device via "Swift Console" you have tcp, udp and more outputs called like this or similar: tcp_1_server. On the filed: "enable_sbp_messages" you will find all messages which are output. Some numbers have a "/50" at the end. This is a divider. This divides the output by the number (on this example 50). 
For IMU set 2304/50 to 2304. For MAG set  2306/50 to 2306. Don`t forget "save to device".
Now it should work:)

pietrocolombo, 2020-08-20:
thanks now it works

> I had the same issue, the support from Swift Nav helped me.
> 
> This is no ROS or driver problem. In the settings of your device via "Swift Console" you have tcp, udp and more outputs called like this or similar: tcp_1_server. On the filed: "enabled_sbp_message " you will find all messages which are output. Some numbers have a "/50" at the end. This is a divider. This divides the output by the number (on this example 50).
> For IMU set 2304/50 to 2304. For MAG set 2306/50 to 2306. Don`t forget "save to device".
> Now it should work:)

thanks now it works

rikba, 2020-08-20:
Thank you @EpicMinister for solving this.

closed 2020-08-20 by rikba