ethz-asl/ethz_piksi_ros #177: ImportError: No module named numba

saeedarabi92, 2020-07-29:
I keep receiving **ImportError: No module named numba** error in my **JetsonTX2**.
**Q**: Do you know how to fix this issue?

```
nvidia@saeed:~/Documents/swift_ros_ws/src/ethz_piksi_ros$ roslaunch piksi_multi_rtk_ros piksi_multi_rover.launch 
... logging to /home/nvidia/.ros/log/81697eda-d1cb-11ea-8629-00044b8c61f2/roslaunch-saeed-11332.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://saeed:43995/

SUMMARY
========

PARAMETERS
 * /piksi/base_station_ip_for_latency_estimation: 192.168.0.1
 * /piksi/base_station_mode: False
 * /piksi/baud_rate: 230400
 * /piksi/broadcast_addr: 192.168.0.255
 * /piksi/broadcast_port: 26078
 * /piksi/debug_mode: False
 * /piksi/driver_verbose: True
 * /piksi/interface: serial
 * /piksi/navsatfix_frame_id: base_link
 * /piksi/publish_raw_imu_and_mag: False
 * /piksi/serial_port: /dev/ttyUSB0
 * /piksi/tcp_addr: 192.168.0.222
 * /piksi/tcp_port: 55555
 * /piksi/use_gps_time: False
 * /piksi/var_rtk_fix: [0.0049, 0.0049, ...
 * /piksi/var_rtk_float: [25.0, 25.0, 64.0]
 * /piksi/var_spp: [25.0, 25.0, 64.0]
 * /piksi/var_spp_sbas: [1.0, 1.0, 1.0]
 * /rosdistro: melodic
 * /rosversion: 1.14.6

NODES
  /
    piksi (piksi_multi_rtk_ros/piksi_multi)

auto-starting new master
process[master]: started with pid [11342]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 81697eda-d1cb-11ea-8629-00044b8c61f2
process[rosout-1]: started with pid [11353]
started core service [/rosout]
process[piksi-2]: started with pid [11360]
/home/nvidia/.local/lib/python2.7/site-packages/quaternion/numba_wrapper.py:21: UserWarning: 

!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
Could not import from numba, which means that some
parts of this code may run MUCH more slowly.  You
may wish to install numba.
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!

  warnings.warn(warning_text)
Traceback (most recent call last):
  File "/home/nvidia/Documents/swift_ros_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/bin/piksi_multi", line 3, in <module>
    from  piksi_multi_rtk_ros import PiksiMulti
  File "/home/nvidia/Documents/swift_ros_ws/devel/lib/python2.7/dist-packages/piksi_multi_rtk_ros/__init__.py", line 34, in <module>
    exec(__fh.read())
  File "<string>", line 1, in <module>
  File "/home/nvidia/Documents/swift_ros_ws/src/ethz_piksi_ros/piksi_multi_rtk_ros/src/piksi_multi_rtk_ros/piksi_multi.py", line 31, in <module>
    from sbp.client.drivers.pyserial_driver import PySerialDriver
  File "/usr/local/lib/python2.7/dist-packages/sbp/client/__init__.py", line 12, in <module>
    from .framer import Framer
  File "/usr/local/lib/python2.7/dist-packages/sbp/client/framer.py", line 11, in <module>
    from ..msg import crc16, SBP, SBP_PREAMBLE
[... 409 lines cut ...]
/usr/local/lib/python2.7/dist-packages/sbp-2.6.5.dist-info/WHEEL
/usr/local/lib/python2.7/dist-packages/sbp-2.6.5.dist-info/top_level.txt
/usr/local/lib/python2.7/dist-packages/sbp2json/__init__.py
/usr/local/lib/python2.7/dist-packages/sbp2json/__init__.pyc
/usr/local/lib/python2.7/dist-packages/sbp2json/__main__.py
/usr/local/lib/python2.7/dist-packages/sbp2json/__main__.pyc
/usr/local/lib/python3.6/dist-packages/sbp
/usr/local/lib/python3.6/dist-packages/sbp-2.6.5.dist-info
/usr/local/lib/python3.6/dist-packages/sbp2json
/usr/share/icons/Humanity/devices/16/usbpendrive_unmount.svg
/usr/share/icons/Humanity/devices/24/usbpendrive_unmount.svg
/usr/share/icons/Humanity/devices/48/usbpendrive_unmount.svg
/usr/share/icons/Tango/16x16/devices/usbpendrive_unmount.png
/usr/share/icons/Tango/22x22/devices/usbpendrive_unmount.png
/usr/share/icons/Tango/24x24/devices/usbpendrive_unmount.png
/usr/share/icons/Tango/32x32/devices/usbpendrive_unmount.png
/usr/share/icons/Tango/scalable/devices/usbpendrive_unmount.svg
/usr/src/linux-headers-4.9.140-tegra-linux_x86_64/kernel-4.9/drivers/target/sbp
/usr/src/linux-headers-4.9.140-tegra-linux_x86_64/kernel-4.9/drivers/target/sbp/Kconfig
/usr/src/linux-headers-4.9.140-tegra-linux_x86_64/kernel-4.9/drivers/target/sbp/Makefile
/usr/src/linux-headers-4.9.140-tegra-ubuntu18.04_aarch64/kernel-4.9/drivers/target/sbp
/usr/src/linux-headers-4.9.140-tegra-ubuntu18.04_aarch64/kernel-4.9/drivers/target/sbp/Kconfig
/usr/src/linux-headers-4.9.140-tegra-ubuntu18.04_aarch64/kernel-4.9/drivers/target/sbp/Makefile

```

nabilahatiqah, 2021-03-03:
did you successfully launch ros file on Jetson tx2?

saeedarabi92, 2021-03-03:
No, I could not solve this problem.

woudie, 2021-07-26:
@saeedarabi92 Hi, did you ever solve this issue? I recently ran into the issue and as far I can tell, it is a convoluted dependency issue where numba is no longer supported by python2 due to its dependency on llvmlite.

nabilahatiqah, 2021-07-28:
just use the numba and llvmlite combination version that support python 2. in my case, i use numba version 0.47 with llvmlite version 0.31

rikba, 2022-05-06:
Please use [piksi_multi_cpp](https://github.com/ethz-asl/ethz_piksi_ros/tree/master/piksi_multi_cpp) with ROS noetic and Piksi Multi Firmware 2.4.16. The Python driver is not maintained and outdated. If you still want to use the Python driver checkout release [v1.10.0](https://github.com/ethz-asl/ethz_piksi_ros/tree/v1.11.0).

closed 2022-05-06 by rikba