ethz-asl/ethz_piksi_ros #181: Compatibility Ubuntu 20.04

SamMuenchow, 2020-09-01:
Hello together,
It seems that the current version is not compatible with ubuntu 20.04. The reason for this seems to be changes in the handling of Python 2 and Python 3. 
Is an update of the software planned?

ebimor, 2021-06-29:
This is also my question. Did you find any information about this? Thanks

rikba, 2022-05-06:
Yes. piksi_multi_cpp is now also running on Ubuntu 20.04 and the latest piksi multi firmware.

Please use [piksi_multi_cpp](https://github.com/ethz-asl/ethz_piksi_ros/tree/master/piksi_multi_cpp) with ROS noetic and Piksi Multi Firmware 2.4.16. The Python driver is not maintained and outdated. If you still want to use the Python driver checkout release [v1.10.0](https://github.com/ethz-asl/ethz_piksi_ros/tree/v1.11.0).

closed 2022-05-06 by rikba