ethz-asl/ethz_piksi_ros #188: piksi_multi_cpp installation not working

SamMuenchow, 2021-01-08:
I have just tried the new installtion guide for the piksi_multi_cpp.

Unforunatly at some point when building with catkin build I get the following error msg:

Errors     << libsbp_ros_msgs:cmake /home/aric/catkin_ws/logs/libsbp_ros_msgs/build.cmake.000.log        
[rospack] Error: package 'libsbp_ros_msgs' not found
CMake Error at /home/aric/catkin_ws/install/share/catkin_simple/cmake/catkin_simple-extras.cmake:150 (add_library):
  Cannot find source file:

    src/conversion.cc

  Tried extensions .c .C .c++ .cc .cpp .cxx .m .M .mm .h .hh .h++ .hm .hpp
  .hxx .in .txx
Call Stack (most recent call first):
  CMakeLists.txt:29 (cs_add_library)

Everything before that is working fine. 

I work with an NVIDIA Xavier with Ubuntu 18.04 and Ros melodic.

rikba, 2021-01-08:
Wow that's some quick feedback.

Can you double check that your catkin workspace is configured to be merge-devel?

```
catkin config --merge-devel
```

We will try to put some more effort into the driver installation and instructions soon.

SamMuenchow, 2021-01-11:
Yes I was trying to get the ethz_piksi_ros running on the Xavier last week but the python version seems not to work with arm64.

The command gives the following output:
 
Workspace configuration appears valid.

Thank you very much for your effort.

rikba, 2021-01-11:
Sorry, what's the output?

SamMuenchow, 2021-01-11:
```
Profile:                     default
Extending:        [explicit] /opt/ros/melodic
Workspace:                   /home/aric/catkin_ws
----------------------------------------------------------------------------
Build Space:        [exists] /home/aric/catkin_ws/build
Devel Space:        [exists] /home/aric/catkin_ws/devel
Install Space:      [exists] /home/aric/catkin_ws/install
Log Space:          [exists] /home/aric/catkin_ws/logs
Source Space:       [exists] /home/aric/catkin_ws/src
DESTDIR:            [unused] None
----------------------------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        merged
----------------------------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
----------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        rqt_rviz rviz_plugin_tutorials librviz_tutorial
----------------------------------------------------------------------------
Workspace configuration appears valid.
```

rikba, 2021-01-11:
Can you try this layout with `Install Space Layout:        None`
```
--------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/melodic
Workspace:                   /home/rik/catkin_ws
--------------------------------------------------------
Build Space:        [exists] /home/rik/catkin_ws/build
Devel Space:        [exists] /home/rik/catkin_ws/devel
Install Space:      [unused] /home/rik/catkin_ws/install
Log Space:          [exists] /home/rik/catkin_ws/logs
Source Space:       [exists] /home/rik/catkin_ws/src
DESTDIR:            [unused] None
--------------------------------------------------------
Devel Space Layout:          merged
Install Space Layout:        None
--------------------------------------------------------
Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
--------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
--------------------------------------------------------
Workspace configuration appears valid.
--------------------------------------------------------
```

I cannot see immediately what's the issue and it will probably take some time until I can provide a more robust solution.

But at least I can try to explain what is happening.

The new CPP driver automatically converts all SBP message types to ROS messages, such that the GNSS receiver will output all available debug information depending on which SBP version is checked out.

The C++ files that do this conversion are [auto-generated during the CMake process](https://github.com/ethz-asl/ethz_piksi_ros/blob/master/libsbp_ros_msgs/CMakeLists.txt#L17-L23). It seems like this auto generation is either failing completely or the autogenerated files are not put in the correct folder.

You can probably get more information by running `catkin build --verbose`.

rikba, 2021-01-13:
Hi Sam

We happened to have the same issue as you have when installing the workspace. We merged a fix, let me know, if this fixes your issue.

SamMuenchow, 2021-01-14:
Hi Rik

I just tried the new version of the code but it did not work either.

But at least the error has changeged. Now i get the following error:
`ImportError: No module named jinja2`

Here is the complet error output:

```
Errors     << libsbp_ros_msgs:cmake /home/aric/catkin_ws/logs/libsbp_ros_msgs/build.cmake.000.log
Traceback (most recent call last):
  File "/home/aric/catkin_ws/src/ethz_piksi_ros-master/libsbp_ros_msgs/src/libsbp_ros_msgs/generator", line 5, in <module>
    import ros_renderer as renderer
  File "/home/aric/catkin_ws/src/ethz_piksi_ros-master/libsbp_ros_msgs/src/libsbp_ros_msgs/ros_renderer.py", line 3, in <module>
    import jinja2
ImportError: No module named jinja2
CMake Error at /home/aric/catkin_ws/install/share/catkin_simple/cmake/catkin_simple-extras.cmake:150 (add_library):
  Cannot find source file:

    src/conversion.cc
```

rikba, 2021-01-14:
I assume that you [did install jinja](https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/install/prepare-jenkins-slave.sh#L13)?

SamMuenchow, 2021-01-14:
Yes I have missed to install jinja.

After rebuilding the project, the build was successful.

Thank you very much for the great support.

closed 2021-01-14 by SamMuenchow

rikba, 2021-01-14:
Thanks for your feedback! :+1: