ethz-asl/ethz_piksi_ros #198: ROS2 support

adamsj-ros, 2021-05-26:
It seems like it's been awhile since a release but not so long since the last commit for this repo. I'm wondering about future plans. Is there any plan to support ROS2?

samuk, 2022-01-02:
I'm also interested in this

MoffKalast, 2022-01-31:
If it compiles on noetic it should be possible to use it through the ROS 1-2 bridge right?

rikba, 2022-05-06:
At the moment we are not on ROS2. However, we have created a ROS noetic compatible driver.

Please use [piksi_multi_cpp](https://github.com/ethz-asl/ethz_piksi_ros/tree/master/piksi_multi_cpp) with ROS noetic and Piksi Multi Firmware 2.4.16. The Python driver is not maintained and outdated. If you still want to use the Python driver checkout release [v1.10.0](https://github.com/ethz-asl/ethz_piksi_ros/tree/v1.11.0).

closed 2022-05-06 by rikba

rikba, 2023-04-18:
There is now an official ROS2 driver by swift-nav: https://github.com/swift-nav/swiftnav-ros2