ethz-asl/ethz_piksi_ros #209: Error on installation ethz_piksi_ros GNSS

tmsp1612, 2022-02-01:
Hello everybody.
I have an issue on the installation of the ROS package ethz_piksi_ros. I have already installed all the dependencies but when I make "catkin_make" this error occur: 
In file included from /home/tmsp/catkin_ws/src/ethz_piksi_ros/piksi_multi_cpp/include/piksi_multi_cpp/receiver/receiver_ros.h:10,
                 from /home/tmsp/catkin_ws/src/ethz_piksi_ros/piksi_multi_cpp/include/piksi_multi_cpp/receiver/receiver_attitude.h:6,
                 from /home/tmsp/catkin_ws/src/ethz_piksi_ros/piksi_multi_cpp/src/receiver/receiver_attitude.cc:1:
/home/tmsp/catkin_ws/src/ethz_piksi_ros/piksi_multi_cpp/include/piksi_multi_cpp/sbp_callback_handler/geotf_handler.h:6:10: fatal error: libsbp_ros_msgs/MsgBasePosEcef.h: No such file or directory
    6 | **#include <### **libsbp_ros_msgs/MsgBasePosEcef.h**>**
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [ethz_piksi_ros/piksi_multi_cpp/CMakeFiles/piksi_multi_cpp.dir/build.make:180: ethz_piksi_ros/piksi_multi_cpp/CMakeFiles/piksi_multi_cpp.dir/src/receiver/receiver_attitude.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/tmsp/catkin_ws/src/ethz_piksi_ros/piksi_multi_cpp/include/piksi_multi_cpp/receiver/receiver_ros.h:10,
                 from /home/tmsp/catkin_ws/src/ethz_piksi_ros/piksi_multi_cpp/include/piksi_multi_cpp/receiver/receiver_base_station.h:9,
                 from /home/tmsp/catkin_ws/src/ethz_piksi_ros/piksi_multi_cpp/src/receiver/receiver_base_station.cc:6:
/home/tmsp/catkin_ws/src/ethz_piksi_ros/piksi_multi_cpp/include/piksi_multi_cpp/sbp_callback_handler/geotf_handler.h:6:10: fatal error: libsbp_ros_msgs/MsgBasePosEcef.h: No such file or directory
    6 | #include <libsbp_ros_msgs/MsgBasePosEcef.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [ethz_piksi_ros/piksi_multi_cpp/CMakeFiles/piksi_multi_cpp.dir/build.make:193: ethz_piksi_ros/piksi_multi_cpp/CMakeFiles/piksi_multi_cpp.dir/src/receiver/receiver_base_station.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:8878: ethz_piksi_ros/piksi_multi_cpp/CMakeFiles/piksi_multi_cpp.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
Invoking "make -j2 -l2" failed

Can someone help me. I have been trying for a week.
Thanks

SamMuenchow, 2022-02-22:
Hi. I think the piksi_multi_cpp is only compatible with catkin build and not with catkin_make.
Furthermore you need to follow the installation manual in the project before building it.

rikba, 2022-05-06:
Please use [piksi_multi_cpp](https://github.com/ethz-asl/ethz_piksi_ros/tree/master/piksi_multi_cpp) with ROS noetic and Piksi Multi Firmware 2.4.16. The Python driver is not maintained and outdated. If you still want to use the Python driver checkout release [v1.10.0](https://github.com/ethz-asl/ethz_piksi_ros/tree/v1.11.0).

closed 2022-05-06 by rikba