ethz-asl/ethz_piksi_ros #220: catkin build problem

dom-lee, 2022-07-13:
Hello, I'm trying to use GNSS RTK receiver with NTRIP.
When I follow the installation in `piksi_multi_cpp`, I keep encounter the below error.

```Traceback (most recent call last):                                                                                         
  File "/home/dongmyeong/Opensource/ethz_piksi/src/ethz_piksi_ros/libsbp_ros_msgs/src/libsbp_ros_msgs/generator", line 56, 
in <module>                                                                                                                
    file_index = yaml.resolve_deps(*yaml.get_files(input_dir))                                                             
  File "/home/dongmyeong/Opensource/ethz_piksi/devel/lib/python3/dist-packages/sbpg/specs/yaml2.py", line 160, in resolve_d
eps                                                                                                                        
    file_index[fname] = read_spec(fname)                                                                                   
  File "/home/dongmyeong/Opensource/ethz_piksi/devel/lib/python3/dist-packages/sbpg/specs/yaml2.py", line 56, in read_spec
    contents = yaml.load(f)
TypeError: load() missing 1 required positional argument: 'Loader'
```

dom-lee, 2022-07-13:
And this is the catkin config
```---------------------------------------------------------------------------
Profile:                     default
Extending:        [explicit] /opt/ros/noetic
Workspace:                   /home/dongmyeong/Opensource/ethz_piksi
---------------------------------------------------------------------------
Build Space:        [exists] /home/dongmyeong/Opensource/ethz_piksi/build
Devel Space:        [exists] /home/dongmyeong/Opensource/ethz_piksi/devel
Install Space:      [unused] /home/dongmyeong/Opensource/ethz_piksi/install
Log Space:          [exists] /home/dongmyeong/Opensource/ethz_piksi/logs
Source Space:       [exists] /home/dongmyeong/Opensource/ethz_piksi/src
DESTDIR:            [unused] None
---------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------------------
Buildlisted Packages:        None
Skiplisted Packages:         None
---------------------------------------------------------------------------
Workspace configuration appears valid.
---------------------------------------------------------------------------```

rikba, 2022-08-22:
@dom-lee did you install the dependencies before building, i.e., `source /opt/ros/noetic/setup.bash
./ethz_piksi_ros/piksi_multi_cpp/install/prepare-jenkins-slave.sh`? In particular 
```
source /opt/ros/noetic/setup.bash
./ethz_piksi_ros/piksi_multi_cpp/install/prepare-jenkins-slave.sh
```
If not, please try this first. Then clean your workspace `catkin clean` and build again.

closed 2022-08-30 by rikba