ethz-asl/ethz_piksi_ros #222: Missing nav topics

Remi-Tortue, 2022-07-29:
Hello,
First thanks for your work !
Im using Piksi Multi 
Package [ethz_piksi_ros](https://github.com/ethz-asl/ethz_piksi_ros) commit 958e3df4e1246ae4171fc6038941683385d4d01e
ROS Noetic
Ubuntu 20.04

Im using piksi_multi_cpp to get rover RTK position.
The RTK work properly in swift console.
But when i launch `roslaunch piksi_multi_cpp base.launch` on base and  `roslaunch piksi_multi_cpp rover.launch` on rover i only get :
/ros/baseline_ned
/ros/receiver_state
/ros/time_ref
/ros/vel_ned

And im supose to get :
/ros/vel_ned
/ros/vel_ned_cov
/ros/pos_enu
/ros/pos_enu_cov
/ros/navsatfix
/ros/receiver_state
/sbp/base_pos_ecef

If anyone have an advice, any help will be appreciate !

chriswsuarez, 2022-08-15:
Hi, has there been any update on this.  I'm seeing the same behavior after recently trying to use `piksi_multi_cpp`

chriswsuarez, 2022-08-15:
I actually just discovered that if I reverted my piksi multi to firmware version 2.4.20 instead of 3.0.11 the ROS node now outputs the desired topics.

Not sure if or when the ROS node will support firmware version 3.0.11

Remi-Tortue, 2022-08-15:
I'm using 2.4.20 too, but have always the issue.
So i used a docker Ubuntu 18.04 ROS Melodic, and with [ethz_piksi_ros](https://github.com/ethz-asl/ethz_piksi_ros) v1.10.0 I get the NavSatFix topic.

rikba, 2022-08-22:
So you are not receiving ros/navsatfix? You should always have this independent of RTK or not... @Remi-Tortue

rikba, 2022-08-30:
Here is a shot why it may have not been working. Make sure in the config of base and rover you are setting the desired device. If you are connecting via ethernet, make sure they are not both starting the same receiver by giving both receivers different IPs.

See

https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/launch/base.launch#L2
https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/launch/rover.launch#L3

closed 2022-08-30 by rikba