ethz-asl/ethz_piksi_ros #228: Navsatfix topic not updating latitude and longitude

mmebc, 2022-09-29:
I am trying to get the live latitude and longitude data through ROS but have been unsuccessful for a while even though I do see the data on the swift console. After running rover.launch I am able to get live data published on other topics like imu raw and gps time but don't see any topics with a navsatfix message on the rover. After running RTK with the base.launch command on the base station I see the following topic on the rover:

`/rover/piksi/position_receiver_0/ros/base_pos_navsatfix`

This has a fixed latitude and longitude value and does not update with time, even when moved around. Is this the topic I need and if yes, what could be the reason it does not update?

I am using the firmware binary PiksiMulti-v3.0.11 since I do not get any data published with v2.4.20. I have also tested with both ROS Noetic and Melodic but the issue persists.

Both the base and rover show the following warnings after running the launch files:
[GeoTF] No TF connection specified.
Failed to convert tow, tow_f to GPS time stamp. Stamping data with ros::Time::now()

One additional issue with the base station is that after calling the ros service `/piksi_multi_cpp_base/base_station_receiver_0/resample_base_position 1000 '/tmp/base_position.txt' false 0.0`
I do not get the topic `/piksi_multi_cpp_base/base_station_receiver_0/position_sampler/position_sampling` pubished.

rikba, 2022-09-30:
Hi there 

`/rover/piksi/position_receiver_0/ros/base_pos_navsatfix` is the base station position that is broadcasted via modem or UDP. This is first of all a good thing you are receiving this and it is constant.

Are you receiving any SBP messages on your rover, e.g.,
```
/moa/piksi/position_receiver_0/sbp/heartbeat
/moa/piksi/position_receiver_0/sbp/pos_llh
```

rikba, 2022-09-30:
Also, how are you connecting your rover, i.e., USB, serial or ethernet?

rikba, 2022-10-01:
Your heartbeat message corresponds to `0b 1 0 000000 00000100 00000001 00000 0 0 0`. According to the [SBP documentation, p.94](https://www.swiftnav.com/resource-files/Swift%20Navigation%20Binary%20Protocol/v3.4.4/Specification/Swift%20Navigation%20Binary%20Protocol%20Specification%20v3.4.4.pdf), your system is healthy, you have an antenna connected, and you are using SBP v4.1 which corresponds to having Piksi Firmware 3.0.11. So in principle you have a successful connection. 

Your pos_llh message on the other hand has `flags: 0`. According to [SBP documentation, p.20](https://www.swiftnav.com/resource-files/Swift%20Navigation%20Binary%20Protocol/v3.4.4/Specification/Swift%20Navigation%20Binary%20Protocol%20Specification%20v3.4.4.pdf) this indicates an `invalid` fix mode. Is the yellow `Pos` LED continuously On (see [Piksi Multi Specifications, p. 6](https://www.swiftnav.com/resource-files/Piksi%20Multi/v2.2/Specification/Piksi%20Multi%20HW%20Specification%20v2.2%20%5B000-534-02-02%5D.pdf))? Are you certainly measuring a fixed position in the Swift Console?

Now, we did not try this driver with Firmware 3.0.11 so it might be that an update in the SBP version may break the driver communication. I can have a look at updating the driver to 3.0.11, but I cannot predict whether this will be next week or in 3 weeks. 

Are you using the blue evaluation board? Can you try the micro USB port instead of the serial cable? I think we did not test the RS232 to USB connection.

rikba, 2022-10-01:
Oh and when you connect it via USB, you can simultaneously open the ROS driver and Swift Console. The ROS driver will tell you which device it connects to, e.g., `/dev/ttyACM0`. Then you can pick the second channel, i.e., `/dev/ttyACM1`, to connect Swift Console.

mmebc, 2022-10-04:
The yellow `Pos` LED does blink at times but moving the antenna higher has fixed it. I see the correct GPS data coming through the Swift console and it gives me centimetre level accuracy with RTK there. 

I tried [this](https://github.com/szenergy/duro_gps_driver) package to see the latitude and longitude update on ROS with my current setup.

I am using the blue evaluation board. Long term I would need to work with only an ethernet cable. I have tried the connection with RS232 to RS232, RS232 to USB and ethernet but haven't tried the micro USB port yet. I will likely test it in a few days.

rikba, 2022-10-12:
Did you test the USB connection?

mmebc, 2022-10-14:
The micro USB connection is working and I am getting an updating navsatfix message on both rover and base station. However, the micro USB cable is the only way I have been able to get it to work and would want it to work with ethernet in the long run.

rikba, 2022-10-19:
Thanks for testing that. If you use the standard configuration of the rover, the messages should be published in the `position_receiver_0` namespace.

To connect the Piksi via ethernet you have to configure the network on your host device and the Piksi accordingly.

For example you can select a dynamic IP and connect the Piksi to a router or you create a static IP on your host computer and set the piksi accordingly. See this setting in the Piksi config: https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/cfg/config_rover_2_4_20.ini#L10-L15

closed 2022-10-19 by rikba