ethz-asl/ethz_piksi_ros #231: Unable to connect to piksi multi

dariojavo, 2023-03-03:
Hi,

I am working with the master branch in ROS Noetic, installation was successful. I have also updated the firmware of my gps to 'v2.4.20' as requested. However, I'm having some issues when try to connect to the piksi multi. Specifically, I'm running:

```sh
roslaunch piksi_multi_cpp config.launch config_type:=rover device_ids:=serial:///dev/ttyUSB0@115200
```
and I'm receiving as output:

```sh
process[piksi_multi_config-2]: started with pid [70467]
[ INFO] [1677808724.206366440]: serial:///dev/ttyUSB0@115200
[ INFO] [1677808724.208390493]: Opened port: /dev/ttyUSB0
[ INFO] [1677808724.215646538]: Opening config: /home/user/Documents/ROS/gps/src/ethz_piksi_ros/piksi_multi_cpp/cfg/config_rover_2_4_20.ini
[ERROR] [1677808756.567476954]: Did not receive write response uart1.mode.SBP
[ WARN] [1677808756.773769080]: Error sbp_process: -2
[ERROR] [1677808757.768319137]: Did not receive write response uart1.baudrate.230400
[ERROR] [1677808759.278557711]: Did not receive write response uart1.flow_control.None
[ERROR] [1677808760.469118301]: Did not receive write response tcp_server0.enabled_sbp_messages.
```
This error goes for all the values in the `config_rover_2_4_20.ini` file.

I have tried increasing the `kSleepMs` in the `settings_io.cc` before and after reading and writing the message, but no changes.

Serial port and baud rate are ok since I'm able to connect using the swift console.

Do you have some thoughts on what could be the issue here?

Thanks!

rikba, 2023-03-14:
Hmm... My first guess would have been the baud rate, also because the device setting is `serial:///dev/ttyUSB0@115200` while the write response is `Did not receive write response uart1.baudrate.230400`. But if you checked and you connect via SwiftConsole, I guess they match.

Did you try to just start [rover.launch](https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/launch/base.launch) or [base.launch](https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/launch/base.launch) to see if it is connecting correctly and outputting any messages, like hearbeat?

dariojavo, 2023-03-14:
Hi @rikba,

Thanks for the response. Yes, I have tried rover.launch but I also was not able to connect via serial. Nevertheless, I was able to connect using micro-usb, I was wondering if it is an issue with serial communication. In #228 they reported a similar problem with serial communication. I have also tried different firmware versions, including the last one, and micro-usb always work, but serial never worked for me.