ethz-asl/ethz_piksi_ros #232: piksi_multi_cpp proper settings for WiFi based communication

chrissunny94, 2023-03-13:
Dear open source community , i have been trying to get the ROS drivers for piksi up and running for the past few days .

- https://github.com/ethz-asl/ethz_piksi_ros/tree/master/piksi_multi_cpp , i have **managed to compile the ROS noetic** version of this driver 
- Also f**lashed the PIKSI hardware** with the prescribed version of **2.4.20 firmware** 

- _Where  to specify the IP address , how  do we pass this argument in the launch file ?_

A stand alone example where all the required configurations are specified would be helpful to many people who are trying to make sense of this .

Best Regards ,
Chris

rikba, 2023-03-14:
Hi there, you can change `device_ids` to the TCP address of your Piksi Multi, e.g., 
```<arg name="device_ids" default="tcp://192.168.222.0:55555"/>```

See [rover.launch](https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/launch/rover.launch#L3)

chrissunny94, 2023-03-16:
I already tried this , but got an error . 

Let me reproduce the error and will post the log here .

Reyshwanth, 2024-05-02:
Hey, I was able to set the right port, I am seeing udp packets from the base station on my wifi network, and they are being recieved at my rover's computer as well. However when I launch the driver , roslaunch piksi_multi_cpp rover.launch, I get the error "Could not bind to UDP socket. Observations will not be received". The port number being read is 26078, which is being read correctly by the file udp_observation_reciever.cc. Why am I unable to bind? please help me out!