ethz-asl/ethz_piksi_ros #234: Build error on ubuntu20.04/noetic. src/conversion.cc not found

andreucm, 2023-07-04:
Hi all, 

1. Given: 
uname -a: Linux maupas 5.15.0-76-generic #83~20.04.1-Ubuntu SMP Wed Jun 21 20:23:31 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux
lsb_release -d: Ubuntu 20.04.6 LTS
rosversion -d: noetic

2. and following [this instructions](https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/README.md#installation)

3. I get the folowing installation error : 

```
Errors     << libsbp_ros_msgs:cmake /home/andreu/dev/workspaces/piksi_gps_ws/logs/libsbp_ros_msgs/build.cmake.000.log             
Traceback (most recent call last):
  File "/home/andreu/dev/workspaces/piksi_gps_ws/src/ethz_piksi_ros/libsbp_ros_msgs/src/libsbp_ros_msgs/generator", line 5, in <module>
    import ros_renderer as renderer
  File "/home/andreu/dev/workspaces/piksi_gps_ws/src/ethz_piksi_ros/libsbp_ros_msgs/src/libsbp_ros_msgs/ros_renderer.py", line 3, in <module>
    import jinja2
  File "/usr/lib/python3/dist-packages/jinja2/__init__.py", line 33, in <module>
    from jinja2.environment import Environment, Template
  File "/usr/lib/python3/dist-packages/jinja2/environment.py", line 15, in <module>
    from jinja2 import nodes
  File "/usr/lib/python3/dist-packages/jinja2/nodes.py", line 23, in <module>
    from jinja2.utils import Markup
  File "/usr/lib/python3/dist-packages/jinja2/utils.py", line 656, in <module>
    from markupsafe import Markup, escape, soft_unicode
ImportError: cannot import name 'soft_unicode' from 'markupsafe' (/home/andreu/.local/lib/python3.8/site-packages/markupsafe/__init__.py)
CMake Error at /home/andreu/dev/workspaces/piksi_gps_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:150 (add_library):
  Cannot find source file:

    src/conversion.cc

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .m .M .mm .h .hh .h++ .hm
  .hpp .hxx .in .txx
Call Stack (most recent call first):
  CMakeLists.txt:35 (cs_add_library)

CMake Error at /home/andreu/dev/workspaces/piksi_gps_ws/devel/share/catkin_simple/cmake/catkin_simple-extras.cmake:150 (add_library):
  No SOURCES given to target: libsbp_ros_msgs
Call Stack (most recent call first):
  CMakeLists.txt:35 (cs_add_library)

CMake Generate step failed.  Build files cannot be regenerated correctly.
cd /home/andreu/dev/workspaces/piksi_gps_ws/build/libsbp_ros_msgs; catkin build --get-env libsbp_ros_msgs | catkin env -si  /usr/bin/cmake /home/andreu/dev/workspaces/piksi_gps_ws/src/ethz_piksi_ros/libsbp_ros_msgs --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/andreu/dev/workspaces/piksi_gps_ws/devel/.private/libsbp_ros_msgs -DCMAKE_INSTALL_PREFIX=/home/andreu/dev/workspaces/piksi_gps_ws/install; cd -

..................................................................................................................................
Failed     << libsbp_ros_msgs:cmake                [ Exited with code 1 ]                                                         
Failed    <<< libsbp_ros_msgs                      [ 1.8 seconds ]                                                                
Abandoned <<< piksi_multi_cpp                      [ Unrelated job failed ]                                                       
Abandoned <<< piksi_multi_interface                [ Unrelated job failed ]                                                       
[build] Summary: 11 of 14 packages succeeded.                                                                                     
[build]   Ignored:   None.                                                                                                        
[build]   Warnings:  3 packages succeeded with warnings.                                                                          
[build]   Abandoned: 2 packages were abandoned.                                                                                   
[build]   Failed:    1 packages failed.                                                                                           
[build] Runtime: 1 minute and 38.8 seconds total.                                                                                 
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

I've seen some related closed issues (https://github.com/ethz-asl/ethz_piksi_ros/issues/230 , https://github.com/ethz-asl/ethz_piksi_ros/pull/212), but do not provide a fix in my case. 

Do you have any hint or comment ? Am I missing something important, or following deprecated instructions ? 

Thanks in advance!

Andreu

andreucm, 2023-07-04:
Fixed , getting inspiration from [here](https://itsourcecode.com/importerror/importerror-cannot-import-name-soft_unicode-from-markupsafe/)

Just doing: `pip install jinja2 --upgrade`

And `catkin build` again.

cse0001, 2024-02-01:
The fundamental issue with your question stems from a dependency management conflict between two software ecosystems. According to the error message, your markupsafe was installed using pip at the path `/home/andreu/.local/lib/python3.8/site-packages/markupsafe/`. The jinja2 that depends on it was installed using apt at the path `/usr/lib/python3/dist-packages/jinja2`. The versions of the two are not compatible. In fact, there is a version of markupsafe in the system that jinja2 correctly depends on, but the Python interpreter prioritizes the markupsafe installed by pip, leading to this issue. There are typically two solutions to this problem: (1) Use pip to uninstall markupsafe, and then reinstall it using apt, or uninstall jinja2 using apt and reinstall it using pip. (2) Use Python's imp module to customize the path and import the markupsafe from the apt path before importing jinja2. An example is as follows:

    import imp
    path = ['/usr/lib/python3/dist-packages']
    fp, pathname, description = imp.find_module('markupsafe', path)
    imp.load_module("markupsafe", fp, pathname, description)

I have developed an automated tool to solve problems like yours and am currently collecting real-world issues. I would really appreciate it if you could reply.

andreucm, 2024-02-02:
Hi @cse0001 , 
thanks for your response. 
As I said, we "solved" the issue upgrading jinja2. 
If I have the opportunity to work with a new fresh machin, i'd try your tool. 
Thanks, 
Andreu

cse0001, 2024-02-04:
@andreucm 
Thanks for your response.