ethz-asl/ethz_piksi_ros #235: No Navsatfix topic showing?

DanRolst, 2023-07-19:
Hi, 

I have a Piksi Multi running on firmware version 3.0.14.
I am trying to use this driver and have installed it and am running the rover launch file.
I can see the lat and long data being published in the sbp topics and i can see its reading all the data correctly. 
However, there is no rostopic being published for navsatfix ?

I am running it on Noetic. Any idea why this topic is not publishing, when i can see the driver is reading the data?

andreucm, 2023-07-20:
Hi, 
I have exactly the same issue. Let me provide additional info about my set up (Ubuntu 20.04, Noetic, firmware 3.0.14)

1. From  SwiftNav console, we set the receiver in simulation mode, turning with a radius of 30m around a given base placed ar ECEF coordinates x,y,z.

2. In the file piksi_multi_cpp/cfg/enu_origin.yaml we set the ECEF x,y,z coordinates of the base  as the origin of the ENU frame.

3. In a terminal we do: `roslaunch piksi_multi_cpp rover.launch device_ids:="tcp://192.168.0.222:55555"`

4. In another terminal , we do: `rostopic list` , with the following result:
```
  /rosout
  /rosout_agg
  /rover/piksi/position_receiver_0/ros/baseline_ned
  /rover/piksi/position_receiver_0/ros/imu
  /rover/piksi/position_receiver_0/ros/receiver_state
  /rover/piksi/position_receiver_0/ros/time_ref
  /rover/piksi/position_receiver_0/ros/vel_ned
  /rover/piksi/position_receiver_0/sbp/age_corrections
  /rover/piksi/position_receiver_0/sbp/base_pos_ecef
  /rover/piksi/position_receiver_0/sbp/baseline_ned
  /rover/piksi/position_receiver_0/sbp/device_monitor
  /rover/piksi/position_receiver_0/sbp/dgnss_status
  /rover/piksi/position_receiver_0/sbp/dops
  /rover/piksi/position_receiver_0/sbp/glo_biases
  /rover/piksi/position_receiver_0/sbp/gps_time
  /rover/piksi/position_receiver_0/sbp/heartbeat
  /rover/piksi/position_receiver_0/sbp/imu_aux
  /rover/piksi/position_receiver_0/sbp/imu_raw
  /rover/piksi/position_receiver_0/sbp/log
  /rover/piksi/position_receiver_0/sbp/measurement_state
  /rover/piksi/position_receiver_0/sbp/network_bandwidth_usage
  /rover/piksi/position_receiver_0/sbp/obs
  /rover/piksi/position_receiver_0/sbp/pos_llh
  /rover/piksi/position_receiver_0/sbp/sv_az_el
  /rover/piksi/position_receiver_0/sbp/thread_state
  /rover/piksi/position_receiver_0/sbp/utc_time
  /rover/piksi/position_receiver_0/sbp/vel_ned
```

The issue is that we do not see any topic publishing the standard [sensor_msgs/NavSatFix](http://docs.ros.org/en/noetic/api/sensor_msgs/html/msg/NavSatFix.html) message. 

Anyone here knows if we are missing to enable some param at ROS level, or at device level ? Or simply this message is not published and we have to get the LLA position from the custom message at topic /sbp/pos_llh ?

Thanks!

Andreu

andreucm, 2023-07-25:
Downgrading firmware to 2.4.20, as stated in the [Readme](https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/README.md). Having the same topics, so NavSatFix does not appear.

andreucm, 2023-07-25:
FIXED: 
I forgot to call first the config launch, as indicated [here](https://github.com/ethz-asl/ethz_piksi_ros/blob/master/piksi_multi_cpp/README.md#basic-usage-base-rover), point 2. 

Sorry about the mess!